NVIDIA/gdrcopy #13: cannot load the driver: Invalid parameters while run ./insert.sh

hanyunfan, 2016-12-14:
# sudo /sbin/insmod gdrdrv/gdrdrv.ko dbg_enabled=0 info_enabled=0
insmod: ERROR: could not insert module gdrdrv/gdrdrv.ko: Invalid parameters

so I tried:

insmod gdrdrv.ko
insmod: ERROR: could not insert module gdrdrv.ko: Invalid parameters

Could you take a look and quick fix it.

drossetti, 2017-02-01:
@hanyunfan that is strange. 
Was that a clean build, i.e. from a pristine directory ?
Which linux kernel ?
Which NVIDIA driver ?
Can you copy here the relevant lines of the system log ? e.g. on RH/Fedora, 'dmesg |tail -20' should be enough.

hanyunfan, 2017-02-01:
@drossetti Thank you for your reply.
It was a clean build on RHCE 7.2 64bit with cuda driver 370.25 and 4x P100 GPUs. 
It's going to be hard to find the dmesg info after a month, but I will try to reproduct it asap.

hanyunfan, 2017-03-13:
The issue was gone.

closed 2017-03-13 by hanyunfan

sharan013, 2017-09-13:
Hi, 
I am hitting the same error. My dmesg error is as follows
[55580.715174] gdrdrv: disagrees about version of symbol nvidia_p2p_get_pages
[55580.715176] gdrdrv: Unknown symbol nvidia_p2p_get_pages (err -22)
[55580.715179] gdrdrv: disagrees about version of symbol nvidia_p2p_put_pages
[55580.715180] gdrdrv: Unknown symbol nvidia_p2p_put_pages (err -22)
[55580.715182] gdrdrv: disagrees about version of symbol nvidia_p2p_free_page_table
[55580.715183] gdrdrv: Unknown symbol nvidia_p2p_free_page_table (err -22)
[66343.379831] gdrdrv: disagrees about version of symbol nvidia_p2p_get_pages
[66343.379836] gdrdrv: Unknown symbol nvidia_p2p_get_pages (err -22)
[66343.379841] gdrdrv: disagrees about version of symbol nvidia_p2p_put_pages
[66343.379842] gdrdrv: Unknown symbol nvidia_p2p_put_pages (err -22)
[66343.379846] gdrdrv: disagrees about version of symbol nvidia_p2p_free_page_table
[66343.379847] gdrdrv: Unknown symbol nvidia_p2p_free_page_table (err -22)
[66394.676111] gdrdrv: disagrees about version of symbol nvidia_p2p_get_pages
[66394.676113] gdrdrv: Unknown symbol nvidia_p2p_get_pages (err -22)
[66394.676117] gdrdrv: disagrees about version of symbol nvidia_p2p_put_pages
[66394.676118] gdrdrv: Unknown symbol nvidia_p2p_put_pages (err -22)
[66394.676120] gdrdrv: disagrees about version of symbol nvidia_p2p_free_page_table
[66394.676121] gdrdrv: Unknown symbol nvidia_p2p_free_page_table (err -22)
[66825.826332] gdrdrv: disagrees about version of symbol nvidia_p2p_get_pages
[66825.826339] gdrdrv: Unknown symbol nvidia_p2p_get_pages (err -22)
[66825.826352] gdrdrv: disagrees about version of symbol nvidia_p2p_put_pages
[66825.826356] gdrdrv: Unknown symbol nvidia_p2p_put_pages (err -22)
[66825.826365] gdrdrv: disagrees about version of symbol nvidia_p2p_free_page_table
[66825.826369] gdrdrv: Unknown symbol nvidia_p2p_free_page_table (err -22)

I am using cuda7.5 on driver version 384 and GeForceGT640 on Ubuntu14.04

drossetti, 2017-09-15:
@sharan013 what if you make the following change?

diff --git a/gdrdrv/Makefile b/gdrdrv/Makefile
index 32e2a9f..a729acf 100644
--- a/gdrdrv/Makefile
+++ b/gdrdrv/Makefile
@@ -45,6 +45,7 @@ REL_MINOR  := $(word 3,$(REL))
 all: module
 
 module:
+       @ echo "Picking NVIDIA driver sources from NVIDIA_SRC_DIR=$(NVIDIA_SRC_DIR). If that does not meet your expectation, you might have a stale driver still around and that might cause problems."
        @ $(MAKE) -C $(KDIR) $(MAKE_PARAMS) M=$(PWD) modules

then try to build and install it:
$ cd gdrdrv
$ make clean all
rm -rf *.o *.ko* *.mod.* .*.cmd Module.symvers modules.order .tmp_versions/ *~ core .depend TAGS 
Picking NVIDIA driver sources from NVIDIA_SRC_DIR=/usr/src/nvidia-387.00/nvidia. If that does not meet your expectation, you might have a stale driver still around and that might cause problems.
make[1]: Entering directory `/usr/src/linux-headers-4.4.0-45-generic'
  CC [M]  /.../gdrcopy/gdrdrv/nv-p2p-dummy.o
  CC [M]  /.../gdrcopy/gdrdrv/gdrdrv.o
  Building modules, stage 2.
...

$ cd ..
$ ./insmod.sh 
INFO: driver major is 238 
INFO: creating /dev/gdrdrv inode
$ lsmod|grep gdr
gdrdrv                 20480  0 
nvidia              14626816  9 gdrdrv,nvidia_modeset,nvidia_uvm

arnoldmcse, 2017-09-18:
@drossetti, that patch pointed out that I was picking up old driver sources. Thanks!