NVIDIA/gdrcopy #18: segfault for copying data buffer of 64-127 byte

chingking, 2017-11-05:
Hi, 

I have seen segfault when copying buffers with size ranging from 64-byte to 127-byte. Followings are the reproducers in our machines.

```
$ /opt/gdrcopy8.0/copybw -s 64
GPU id:0 name:Tesla K40c PCI domain: 0 bus: 2 device: 0
selecting device 0
testing size: 64
rounded size: 65536
device ptr: b05a40000
bar_ptr: 0x7f43d9223000
info.va: b05a40000
info.mapped_size: 65536
info.page_size: 65536
page offset: 0
user-space pointer:0x7f43d9223000
BAR writing test, size=64 offset=0 num_iters=10000
BAR1 write BW: 457.923MB/s
BAR reading test, size=64 offset=0 num_iters=100
Segmentation fault

$dmesg
...
[2689239.364734] copybw[5308]: segfault at 2846000 ip 00007f43d8ecd06c sp 00007fff23b939e0 error 6 in libgdrapi.so.1.2[7f43d8ecb000+3000]
```

```
$ /opt/gdrcopy8.0/copybw -s 64
GPU id:0 name:Tesla K80 PCI domain: 0 bus: 5 device: 0
GPU id:1 name:Tesla K80 PCI domain: 0 bus: 6 device: 0
selecting device 0
testing size: 64
rounded size: 65536
device ptr: 2304fc0000
bar_ptr: 0x2acc78311000
info.va: 2304fc0000
info.mapped_size: 65536
info.page_size: 65536
page offset: 0
user-space pointer:0x2acc78311000
BAR writing test, size=64 offset=0 num_iters=10000
BAR1 write BW: 722.593MB/s
BAR reading test, size=64 offset=0 num_iters=100
Segmentation fault (core dumped)

$dmesg
...
[2614698.728292] copybw[32532]: segfault at 2acc78321000 ip 00002acc78459018 sp 00007ffd51c16b10 error 4 in libgdrapi.so.1.2[2acc78457000+3000]
```

Do you have any idea what could be happening here?

Thanks,

drossetti, 2017-11-18:
@chingking which branch were you testing on?

chingking, 2017-11-18:
@drossetti - I was using `master`. Should I try branch `devel`?

drossetti, 2017-11-18:
@chingking it was an oversight, look at d671b4d

drossetti, 2017-11-18:
@chingking just integrated fix in master as well, could you verify plz?

closed 2017-11-18 by drossetti

chingking, 2017-11-18:
@drossetti - It is working! Thanks for the fix.

commit d671b4d7 (2017-11-18): fix unroll loop in the aligned sse41 case
fixes #18
commit 45768897 (2017-11-18): fix unroll loop in the aligned sse41 case
fixes #18