NVIDIA/gdrcopy #19: a question about CPU mappings

JsonZhangAA, 2017-11-06:
Hi,
    I use these same APIs to create perfectly valid CPU mappings of two GPU memory.Like this:
     `   //------dev_b pin buff----------------------------------------------------------
      	    	  unsigned int flag_b;
      	     	  cuPointerSetAttribute(&flag_b, CU_POINTER_ATTRIBUTE_SYNC_MEMOPS, dev_b);
      	     	  gdr_mh_t mh_b;
      	     	  gdr_t g_b=gdr_open();
      	     	  ASSERT_NEQ(g_b, (void*)0);
      	     	  gdr_pin_buffer(g_b,dev_b,sizeof(int),0,0,&mh_b);
      	     	   void * bar_ptr_b=NULL;
      	     	   ASSERT_EQ(gdr_map(g_b, mh_b, &bar_ptr_b, sizeof(int)), 0);
      	     	   gdr_info_t info_b;
      	     	   	  gdr_get_info(g_b,mh_b,&info_b);
      	     	   	  int off_b=dev_b-info_b.va;
      	     	   	  cout<<"off_b:"<<off_b<<endl;
      	     	   	  uint32_t * buf_ptr_b=(uint32_t *)((char *)bar_ptr_b+off_b);
      	     	   	  cout<<"buf_ptr_b:"<<buf_ptr_b<<endl;
      	     	   	  //---------------------------------------------------------------------------------------
       //-------dev_a pin buff------------------------------------------------------
	  unsigned int flag;
      cuPointerSetAttribute(&flag, CU_POINTER_ATTRIBUTE_SYNC_MEMOPS, dev_a);
	  gdr_mh_t mh;
	  gdr_t g=gdr_open();
	  ASSERT_NEQ(g, (void*)0);
	  gdr_pin_buffer(g, dev_a, N*sizeof(int), 0, 0, &mh);
	  void * bar_ptr=NULL;
	  ASSERT_EQ(gdr_map(g, mh, &bar_ptr, N*sizeof(int)), 0);
	  gdr_info_t info;
	  gdr_get_info(g,mh,&info);
	  int off=dev_a-info.va;
	  cout<<"off_a:"<<off<<endl;
	  uint32_t * buf_ptr=(uint32_t *)((char *)bar_ptr+off);
	  cout<<"buf_ptr:"<<buf_ptr<<endl;`
But it failed.And I find that It's related to the order of a and b, the previous success, the subsequent failure.
       Do you have any idea what could be happening here?

Thanks,

drossetti, 2017-11-18:
@JsonZhangAA why are you calling gdr_open() twice?

drossetti, 2017-11-27:
closing this as not a bug.

closed 2017-11-27 by drossetti