NVIDIA/gdrcopy #25: add autotuning support

drossetti, 2017-12-06:
optimized memcpy implementations should be chosen at run-time during a tuning phase, possibly in gdr_open()

maddyscientist, 2017-12-06:
Curious: what dimensions are you going to be tuning over here in the autotuner?

drossetti, 2018-06-20:
@maddyscientist that is a good question. I am not expecting a dependency on the buffer size, but I might be wrong.

hongbilu, 2023-08-21:
@drossetti BTW, is there any calculation formula, otherwise that would depend on experimental values on kinds of HW configuration

drossetti, 2023-08-21:
@hongbilu any performance model would be HW dependent inherently, so it would involve maintaining a database of FOMs for each platform. That is why I was proposing a run-time autotuning phase instead.

hongbilu, 2023-08-22:
@drossetti that would be a big work and cpu's work frequency or workload also need to be considered in theory. Experiments show that cpu's work frequency is a key influence factor