NVIDIA/gdrcopy #30: nvidia_p2p_get_pages() failed

mdclemen, 2018-03-07:
I just installed gdrcopy on my machine (Ubuntu 16.04.4 LTS (GNU/Linux 4.4.0-116-generic x86_64)) using CUDA 7.5, V7.5.17 (NVIDIA driver version 367.27), with an NVIDIA Tesla K20m GPU. After trying to run $ ./validate, the following error was printed in dmesg:
gdrdrv:nvidia_p2p_get_pages(va=704fe0000 len=327680 p2p_token=0 va_space=0) failed [ret = -22]

Does anyone have any bright ideas on why I can't do GPUDirect RDMA? Thanks.

mdclemen, 2018-03-07:
Turns out, my driver and CUDA combination were not compatible.

closed 2018-03-07 by mdclemen

Jeongseob, 2019-08-01:
Hi @mdclemen,

I am also facing the same issue, but I couldn't find which the driver and CUDA version incur the error. I am wondering you did run `gdrcopy` with the lower version of driver or CUDA than the version specified in README.md at that time?
  
In my case, I am using following driver and CUDA software on two RTX2080-ti GPUs.
- Driver Version: 430.14 
- CUDA Version: 10.2

If you have any suggestion, it would be great. 

Thanks,
Jeongseob

Jeongseob, 2019-08-02:
Hi @drossetti,

I do not have permission to reopen this issue. So, I just wanted you in this loop. Can you have any suggestions to resolve this issue, failing nvidia_p2p_get_pages() while running the `validate` binary?

drossetti, 2019-08-02:
@jeongseob unfortunately your GeForce card does not support GPUDirect RDMA feature. Please read carefully the prerequisites section of the README.md file.

Jeongseob, 2019-08-02:
Ah, I just assumed that RTX2080 (Turing architecture) supports GPUDirect RDMA, but it is not.

Thanks!

nevion, 2020-12-18:
@drossetti  is there any device property query that indicates a card supports RDMA or not?

drossetti, 2023-10-24:
for the records, sometimes during CUDA 11.6, the attribute below was added:
[CU_DEVICE_ATTRIBUTE_GPU_DIRECT_RDMA_SUPPORTED](https://docs.nvidia.com/cuda/cuda-driver-api/group__CUDA__TYPES.html#group__CUDA__TYPES_1gge12b8a782bebe21b1ac0091bf9f4e2a31d37205b1c55a2ef056e240fed6aca13): Device supports GPUDirect RDMA APIs, like nvidia_p2p_get_pages (see https://docs.nvidia.com/cuda/gpudirect-rdma for more information)