NVIDIA/gdrcopy #34: Failed tests (validate and copybw) with latest Mellanox driver and kernel

junjieqian, 2018-03-20:
Hi, we found the tests break the machine with CUDA-9, Nvidia driver (390.25), latest Mellanox driver (v4_3-1_0_1_0) and Linux kernel (4.14.11). The installation of the gdrcopy finishes successfully, but the machine freezes and then reboots when running the tests.
any suggestions? Thanks

lwdhw1987, 2018-04-23:
Hi, I got same problem on my machine, Have you addressed  this ? Is there any reason for this problem.

drossetti, 2018-06-20:
did you solve this problem?

junjieqian, 2018-06-21:
Hi, the issue still happens. The library is installed with ubuntu16.04 container and on host with coreOS (kernel version: 4.14.11). It is kind of impossible to grab any dump info, as the machine would freeze and then entering reboot loop.

drossetti, 2018-06-21:
There are many ways to debug a kernel crash, for example see https://help.ubuntu.com/lts/serverguide/kernel-crash-dump.html

drossetti, 2018-06-21:
besides, we need:
- a description of the HW, server/workstation make and model, cards installed
- a list of SW component
- a clear list of steps to reproduce

drossetti, 2020-06-09:
@junjieqian I'll need an update on this or feel free to close this issue otherwise.

closed 2020-06-17 by junjieqian