NVIDIA/gdrcopy #35: Failed to make install

seiriosPlus, 2018-03-22:
I get this exception:

ln -sf libgdrapi.so.1.2 libgdrapi.so.1
ln -sf libgdrapi.so.1 libgdrapi.so
cd gdrdrv; \
        /usr/bin/make64 
make64[1]: Entering directory `/home/users/tangwei12/gdrcopy-master/gdrdrv'
Picking NVIDIA driver sources from NVIDIA_SRC_DIR=/usr/src/nvidia-linux-x86_64-390.12/kernel/nvidia. If that does not meet your expectation, you might have a stale driver still around and that might cause problems.
make64[2]: Entering directory `/home/users/tangwei12/linux-4-14'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /home/users/tangwei12/gdrcopy-master/gdrdrv/nv-p2p-dummy.o
  CC [M]  /home/users/tangwei12/gdrcopy-master/gdrdrv/gdrdrv.o
  Building modules, stage 2.
  MODPOST 2 modules
FATAL: /home/users/tangwei12/gdrcopy-master/gdrdrv/gdrdrv.o is truncated. sechdrs[i].sh_offset=7089075323386670592 > sizeof(*hrd)=64
make64[3]: *** [__modpost] Error 1
make64[2]: *** [modules] Error 2
make64[2]: Leaving directory `/home/users/tangwei12/linux-4-14'
make64[1]: *** [module] Error 2
make64[1]: Leaving directory `/home/users/tangwei12/gdrcopy-master/gdrdrv'
make64: *** [driver] Error 2

OS: centOS 6.3 (4.14.18)
CUDA: 9
Driver Version: 390.12

drossetti, 2018-04-23:
It looks like your kernel directory does not contain a build and/or is correctly configured.

drossetti, 2018-06-20:
this is not a gdrcopy bug
closing

closed 2018-06-20 by drossetti