NVIDIA/gdrcopy #38: some questions to use this repo

shinespeciall, 2018-04-19:
1. is there any documents to help me know about how to use code files in this repo in my own project?
2. is it possible to make all the function into a dll or lib file for convenience?
3. so i can use this to make screenshots or video streaming for games run by NVIDIA Geforce GPU?

drossetti, 2018-04-23:
#1 you can refer to the GPUdirect RDMA docs ...
#2 where we mention that RDMA only works on Linux.
#3 I don’t believe so. RDMA is a compute technology, while games usually use graphics. The level of interop between the two technologies is described in the docs.

shinespeciall, 2018-04-24:
thanks for help, but anther question, I want to make a program on windows to video streaming and XInput remote emulating for dolphin emulator, can it be possible and convenient to video streaming in high quality by GPU APIs or Windows APIs ? what technique is generally used for video streaming?

drossetti, 2018-06-20:
sorry that is not in my area of expertise
closing

closed 2018-06-20 by drossetti