NVIDIA/gdrcopy #44: gdr_map returns  -EAGAIN

bureddy, 2018-11-16:
```

 
[1218757.588122] gdrdrv:mmap start=0x7f45e83c3000 size=196608 off=0xc31d2952
[1218757.588123] gdrdrv:range start with p=0 vaddr=7f45e83c3000 page_paddr=3838082a0000
[1218757.588125] gdrdrv:non-contig p=1 prev_page_paddr=3838082a0000 cur_page_paddr=3838084b0000
[1218757.588127] gdrdrv:mapping p=1 entries=1 offset=0 len=65536 vaddr=7f45e83c3000 paddr=3838082a0000
[1218757.588128] gdrdrv:mmaping phys mem addr=0x3838082a0000 size=65536 at user virt addr=0x7f45e83c3000
[1218757.588129] gdrdrv:is_cow_mapping is FALSE
[1218757.588138] gdrdrv:range start with p=1 vaddr=7f45e83d3000 page_paddr=3838084b0000
[1218757.588139] gdrdrv:mapping p=3 entries=2 offset=0 len=131072 vaddr=7f45e83d3000 paddr=3838084b0000
[1218757.588141] gdrdrv:mmaping phys mem addr=0x3838084b0000 size=131072 at user virt addr=0x7f45e83d3000
[1218757.588141] gdrdrv:is_cow_mapping is FALSE
[1218757.588146] gdrdrv:track_pfn_remap failed :-22
[1218757.588150] gdrdrv:error in remap_pfn_range() ret:-22
[1218757.588151] gdrdrv:error -11 in gdrdrv_mmap_phys_mem_wcomb

```

bureddy, 2018-11-16:
io_remap_pfn_range-> track_pfn_remap return -EINVAL

drossetti, 2019-01-08:
@bureddy is this an issue?

drossetti, 2019-05-01:
@bureddy I'm closing this as not a bug

closed 2019-05-01 by drossetti