NVIDIA/gdrcopy #47: Need -lrt in Makefile

tonycurtis, 2019-02-08:
```shell
$ make
...
...
/usr/bin/ld: copybw.o: undefined reference to symbol 'clock_gettime@@GLIBC_2.2.5'
/usr/bin/ld: note: 'clock_gettime@@GLIBC_2.2.5' is defined in DSO /lib64/librt.so.1 so try adding it to the linker command line
/lib64/librt.so.1: could not read symbols: Invalid operation
collect2: error: ld returned 1 exit status
make: *** [copybw] Error 1
```

Had to add "-lrt" to LIBS in Makefile:15

drossetti, 2019-02-08:
@tonycurtis which OS/distro were you using?

tonycurtis, 2019-02-08:
> On Feb 8, 2019, at 6:26 PM, Davide Rossetti <notifications@github.com> wrote:
> 
> @tonycurtis <https://github.com/tonycurtis> which OS/distro were you using?
> 

Old one: Fedora 18

tony

drossetti, 2019-05-01:
Eventually I'll need to switch to autoconf/automake to take care of that.

pakmarkthub, 2019-05-30:
@tonycurtis We are switching to autoconf. Can you try this branch (https://github.com/pakmarkthub/gdrcopy/tree/dev-autotools) out and see whether it addresses your issue or not?

tonycurtis, 2019-05-30:
Sure, will give it a go...

tonycurtis, 2019-05-30:
Looks like it works, was able to compile gdrdrv

pakmarkthub, 2019-05-30:
Thank you for the confirmation.

Let me double-check just in case. Did you pass "--enable-test" to ./configure? Right now, copybw is not compiled by default since it is a test program.

tonycurtis, 2019-05-30:
Oh, no.  This is what configure gives me with `--enable-test`

```shell
configure: Test is enabled
checking for cuMemAlloc in -lcuda... yes
checking for cudaMalloc in -lcudart... no
configure: error: in `/usr/local/src/gdrcopy/autoconf':
configure: error: libcudart not found
See `config.log' for more details
```

There's nothing in the `configure --help` output about where CUDA is

pakmarkthub, 2019-05-30:
It cannot find cuda. Could you make sure that LD_LIBRARY_PATH, LIBRARY_PATH, C_PATH points to your cuda folder and run the script again?

tonycurtis, 2019-05-30:
LD_LIBRARY_PATH and CPATH were set, but not LIBRARY_PATH.  The latter is required.  I know this is a matter of autoconf debate but being able to set this via `--with-cuda` would be nice.

tonycurtis, 2019-05-30:
How do I run the tests?

```shell
eureka[1](/usr/local/src/gdrcopy/autoconf) make test
make: *** No rule to make target `test'.  Stop.
eureka[1](/usr/local/src/gdrcopy/autoconf) make tests
make: Nothing to be done for `tests'.
eureka[1](/usr/local/src/gdrcopy/autoconf) make check
Making check in src
make[1]: Entering directory `/usr/local/src/gdrcopy/autoconf/src'
Making check in gdrdrv
make[2]: Entering directory `/usr/local/src/gdrcopy/autoconf/src/gdrdrv'
make[2]: *** No rule to make target `check'.  Stop.
make[2]: Leaving directory `/usr/local/src/gdrcopy/autoconf/src/gdrdrv'
make[1]: *** [check-recursive] Error 1
make[1]: Leaving directory `/usr/local/src/gdrcopy/autoconf/src'
make: *** [check-recursive] Error 1
```

pakmarkthub, 2019-05-30:
Thank you for the feedback. Let me improve the build script and contact you back again. Surprisingly, the machines we have tested so far do not show this problem.

tonycurtis, 2019-05-30:
It might be the much older Fedora.  I can give you an account for testing if you like.

pakmarkthub, 2019-06-06:
@tonycurtis Thank you for waiting. I eventually have time to come back to this issue.

Can you give it another try? If the issue is still there, can I ask for an account so that I can debug on your machine?

P.S. I also add "--with-cuda" as you requested.

tonycurtis, 2019-06-07:
That looks good, thanks.  You could presumably simplify autogen.sh to something like e.g.

[UCX](https://github.com/openucx/ucx/blob/master/autogen.sh)

pakmarkthub, 2019-06-07:
Thank you for the feedback as well.

I will close this issue after the PR gets merged upstream.

pakmarkthub, 2019-08-22:
The fix has been integrated to the master branch. Closing this issue.

closed 2019-08-22 by pakmarkthub