NVIDIA/gdrcopy #48: gdrcopy-devel RPM won't install

mrobbert, 2019-03-26:
I have built the gdrcopy RPMS from the build_packages.sh script in the source tree and I've found that I'm unable to install the gdrcopy-devel package because it is missing a required dependency.

Error: Package: gdrcopy-devel-1.3-2.x86_64 (/gdrcopy-devel-1.3-2.x86_64)
           Requires: libgdrapi.so.1()(64bit)
The file listed was installed by the gdrcopy RPM, but that library didn't get listed as being provided by the RPM. 
If this works for other users then I may have messed something up in my environment, but if not it is probably a bug in the spec file that should get fixed. 
Either way I'm willing to put some work into figuring it out, but I wanted to know which side of the problem to focus on.

drossetti, 2019-05-01:
right, it is a problem in the spec file.
In the past, I have not found a way to write it in such a way to work well for both RH6 and RH7.
If you figure it out, feel free to share the fix via a PR.

drossetti, 2019-05-01:
BTW you can force the installation of the gdrcopy-devel package. 
Everything works because libgdrapi.so.1 is actually installed by gdrcopy.

mrobbert, 2019-05-09:
I have created PR #53 that I believe fixes this issue.

drossetti, 2019-05-28:
@pakmarkthub can you help with this?

drossetti, 2020-06-09:
this issue has been actually been fixed, see https://github.com/NVIDIA/gdrcopy/pull/53#issuecomment-531387444

closed 2020-06-09 by drossetti