NVIDIA/gdrcopy #72: Issues with gdr driver

afernandezody, 2019-07-30:
Hello,
I'm running into some issues while trying to use gdrcopy in a MPI environment. I have CUDA 10.1 (418.67) and the error reads:
`GDRCOPY library "libgdrapi.so" unable to open GDR driver, is gdrdrv.ko loaded?
`
I'm new to gdrcopy and don't really know what this means. After installing gdrcopy, I performed the suggested validations that read OK to me:
```
 ./validate
buffer size: 327680
off: 0
check 1: MMIO CPU initialization + read back via cuMemcpy D->H
check 2: gdr_copy_to_bar() + read back via cuMemcpy D->H
check 3: gdr_copy_to_bar() + read back via gdr_copy_from_bar()
check 4: gdr_copy_to_bar() + read back via gdr_copy_from_bar() + 5 dwords offset
check 5: gdr_copy_to_bar() + read back via gdr_copy_from_bar() + 11 bytes offset
warning: buffer size 1763323920 is not dword aligned, ignoring trailing bytes
unampping
unpinning
 ./copybw
GPU id:0 name:Tesla K80 PCI domain: 0 bus: 0 device: 4
selecting device 0
testing size: 131072
rounded size: 131072
device ptr: 403960000
bar_ptr: 0x7f0395d5c000
info.va: 403960000
info.mapped_size: 131072
info.page_size: 65536
page offset: 0
user-space pointer:0x7f0395d5c000
writing test, size=131072 offset=0 num_iters=10000
write BW: 9437.68MB/s
reading test, size=131072 offset=0 num_iters=100
read BW: 356.296MB/s
unmapping buffer
unpinning buffer
closing gdrdrv
```
Any suggestions on how to proceed or what am I missing? Thanks.

drossetti, 2019-07-30:
Most probably the gdrdrv kernel module is either not loaded or not accessible (eg protection flags on the /dev/gdrdrv inode, container environment, etc) on the cluster nodes where your MPI job is run.

afernandezody, 2019-07-31:
Thanks. That sounds the right diagnosis. This is the first time I hear about protection flags (in this context) and couldn't find information about it. Is there any data or parameter that I should be reading to provide a more complete description of the system configuration?

pakmarkthub, 2019-07-31:
Can I ask you to run two commands and post the results?
1. `ls -alh /dev/gdrdrv`
2. `lsmod | grep gdrdrv`

You may want to try `sudo ./insmod.sh` to reload the driver. The script is in the top directory.

afernandezody, 2019-07-31:
After reloading the driver, I get:
ls -alh /dev/gdrdrv ---> crw-rw-rw-. 1 root root 235, 0 Jul 31 19:31 /dev/gdrdrv
lsmod | grep gdrdrv ---> 
gdrdrv                 17352  0
nvidia              17901608  3 nv_peer_mem,gdrdrv,nvidia_modeset

pakmarkthub, 2019-07-31:
Can you use gdrcopy now?

afernandezody, 2019-08-01:
No as I'm still getting the same error message. I suspect that the issue is some interference (maybe the protection flag that @drossetti mentioned) as the module has always been loaded during the tests. I could try two things:

1. Have a similar configuration but w/o loading the NVIDIA_peer_memory
2. Create a similar configuration but in a different environment (instance). However, this will take me a few hours as I have to build a few pieces from source   
 
Thanks.

pakmarkthub, 2019-08-01:
I believe that @drossetti meant `ls -alh /dev/gdrdrv ---> crw-rw-rw-. 1 root root 235, 0 Jul 31 19:31 /dev/gdrdrv`. Your user needs read and write permissions to /dev/gdrdrv. In your case, the flags are set for all users, so that should not be your issue.

You mentioned that the error occurs when you ran MPI. Is it possible that gdrdrv is not loaded properly on some nodes?

If you have verified that and the error is still persisted, can I ask you to enable the debug mode and post the logs here?
1. Edit insmod.sh. Change line 28 `sudo /sbin/insmod gdrdrv/gdrdrv.ko dbg_enabled=0 info_enabled=0` to `sudo /sbin/insmod gdrdrv/gdrdrv.ko dbg_enabled=1 info_enabled=1`.
2. On another terminal, run `dmesg -w`. You should see more lines show up after you run the following steps.
3. Do `sudo ./insmod.sh`.
4. `export GDRCOPY_ENABLE_LOGGING=1`
5. `export GDRCOPY_LOG_LEVEL=1`
6. Run your MPI. Remember to use `mpirun -x GDRCOPY_ENABLE_LOGGING -x GDRCOPY_LOG_LEVEL <the rest of the command>` so that the environment variables are properly passed to your mpi environment.
7. Post the log from gdrcopy.
8. Post the log from dmesg. Only what shows up after the step 2 is fine.

afernandezody, 2019-08-02:
Sorry about the delay as I'm troubleshooting several things. gdrcopy is running in a single node (1 vCPU + 1 GPU). I believe that the gdr error message can be avoided by including the flag MV2_GPUDIRECT_GDRCOPY_LIB=/home/ody/gdrcopy/libgdrapi.so (not sure why gdrcopy didn't install at /opt as specified but that is a different question). The dmesg output now reads:
[ 2058.859349] gdrdrv:minor=0
[ 2060.863156] gdrdrv:closing
I intend on building a new system tomorrow and will re-test everything. Accordingly, I'll post any update asap and will close the ticket if the issue was simply caused by not specifying the library at the time of initializing the MPI job. Thanks.

afernandezody, 2019-08-05:
It seems that declaring the flag MV2_GPUDIRECT_GDRCOPY_LIB as the specific subdirectory & library solves the problem ($LD_LIBRARY_PATH had the subsirectory so I'm unsure what was originating the error). Another issue came up while building UCX but I'm closing this ticket and will open a different one if necessary. Thank you for the help.

closed 2019-08-05 by afernandezody