NVIDIA/gdrcopy #75: Errors with libgdrapi.so.1.2 while building

afernandezody, 2019-08-07:
Hello,
I've run into the following error message while building gdrcopy-v1.3 (it doesn't happen with the master branch):
```
sudo make CUDA=/usr/local/cuda-10.1 all install
make: execvp: ./config_arch: Permission denied
echo "GDRAPI_ARCH="
GDRAPI_ARCH=
cd gdrdrv; \
make
make[1]: Entering directory `/home/ody/gdrcopy-1.3/gdrdrv'
Picking NVIDIA driver sources from NVIDIA_SRC_DIR=/usr/src/nvidia-418.67/nvidia. If that does not meet your expectation, you might have a stal                                   e driver still around and that might cause problems.
make[2]: Entering directory `/usr/src/kernels/3.10.0-957.27.2.el7.x86_64'
  Building modules, stage 2.
  MODPOST 2 modules
make[2]: Leaving directory `/usr/src/kernels/3.10.0-957.27.2.el7.x86_64'
make[1]: Leaving directory `/home/ody/gdrcopy-1.3/gdrdrv'
g++ -O2 -I /usr/local/cuda-10.1/include -I gdrdrv/ -I /usr/local/cuda-10.1/include -D GDRAPI_ARCH= -L /usr/local/cuda-10.1/lib64 -L /usr/local                                   /cuda-10.1/lib -L /usr/lib64/nvidia -L /usr/lib/nvidia -L /usr/local/cuda-10.1/lib64   -o basic basic.o libgdrapi.so.1.2 -lcudart -lcuda -lpth                                   read -ldl
libgdrapi.so.1.2: undefined reference to `memcpy_cached_store_sse'
libgdrapi.so.1.2: undefined reference to `memcpy_uncached_store_avx'
libgdrapi.so.1.2: undefined reference to `memcpy_cached_store_avx'
libgdrapi.so.1.2: undefined reference to `memcpy_uncached_store_sse'
libgdrapi.so.1.2: undefined reference to `memcpy_uncached_load_sse41'
collect2: error: ld returned 1 exit status
make: *** [basic] Error 1
```
The hardware is a virtualized environment with an Intel(R) Xeon(R) CPU @ 2.30GHz and a 00:04.0 3D controller: NVIDIA Corporation GK210GL [Tesla K80] (rev a1) ) GPU. Thanks.

drossetti, 2019-08-21:
That `permission denied` error is the problem, most probably.
Could you please run `make all` without sudo? The directory you are building is probably a network filesystem where local root has not enough privileges.

afernandezody, 2019-08-22:
Hi @drossetti,
I put this on hold and was working on some other projects. I tried again today but ran into the same issue while compiling as a regular user:
```
$ make CUDA=/usr/local/cuda-10.1 all
make: execvp: ./config_arch: Permission denied
echo "GDRAPI_ARCH="
GDRAPI_ARCH=
cc -O2 -fPIC -I /usr/local/cuda-10.1/include -I gdrdrv/ -I /usr/local/cuda-10.1/include -D GDRAPI_ARCH=  -c -o gdrapi.o gdrapi.c
cc -shared -Wl,-soname,libgdrapi.so.1 -o libgdrapi.so.1.2 gdrapi.o
ldconfig -n /home/odyhpc/gdrcopy-1.3
ln -sf libgdrapi.so.1.2 libgdrapi.so.1
ln -sf libgdrapi.so.1 libgdrapi.so
cd gdrdrv; \
make
make[1]: Entering directory `/home/odyhpc/gdrcopy-1.3/gdrdrv'
Picking NVIDIA driver sources from NVIDIA_SRC_DIR=/usr/src/nvidia-418.67/nvidia. If that does not meet your expectation, you might have a stale driver still around and that might cause problems.
make[2]: Entering directory `/usr/src/kernels/3.10.0-957.27.2.el7.x86_64'
  CC [M]  /home/odyhpc/gdrcopy-1.3/gdrdrv/nv-p2p-dummy.o
  CC [M]  /home/odyhpc/gdrcopy-1.3/gdrdrv/gdrdrv.o
  Building modules, stage 2.
  MODPOST 2 modules
  CC      /home/odyhpc/gdrcopy-1.3/gdrdrv/gdrdrv.mod.o
  LD [M]  /home/odyhpc/gdrcopy-1.3/gdrdrv/gdrdrv.ko
  CC      /home/odyhpc/gdrcopy-1.3/gdrdrv/nv-p2p-dummy.mod.o
  LD [M]  /home/odyhpc/gdrcopy-1.3/gdrdrv/nv-p2p-dummy.ko
make[2]: Leaving directory `/usr/src/kernels/3.10.0-957.27.2.el7.x86_64'
make[1]: Leaving directory `/home/odyhpc/gdrcopy-1.3/gdrdrv'
g++ -O2 -I /usr/local/cuda-10.1/include -I gdrdrv/ -I /usr/local/cuda-10.1/include -D GDRAPI_ARCH=  -c -o basic.o basic.cpp
g++ -O2 -I /usr/local/cuda-10.1/include -I gdrdrv/ -I /usr/local/cuda-10.1/include -D GDRAPI_ARCH= -L /usr/local/cuda-10.1/lib64 -L /usr/local/cuda-10.1/lib -L /usr/lib64/nvidia -L /usr/lib/nvidia -L /usr/local/cuda-10.1/lib64   -o basic basic.o libgdrapi.so.1.2 -lcudart -lcuda -lpthread -ldl
libgdrapi.so.1.2: undefined reference to `memcpy_cached_store_sse'
libgdrapi.so.1.2: undefined reference to `memcpy_uncached_store_avx'
libgdrapi.so.1.2: undefined reference to `memcpy_cached_store_avx'
libgdrapi.so.1.2: undefined reference to `memcpy_uncached_store_sse'
libgdrapi.so.1.2: undefined reference to `memcpy_uncached_load_sse41'
collect2: error: ld returned 1 exit status
make: *** [basic] Error 1

```
I'm just following the same steps as for compiling the newer gdrcopy version. Just in case it's of any use, the subdirectory and permissions look like:

```
$ ls -l
total 144
-rw-rw-r--. 1 odyhpc odyhpc  2615 Aug  6 23:40 basic.cpp
-rw-rw-r--. 1 odyhpc odyhpc  4296 Aug 22 20:47 basic.o
-rw-rw-r--. 1 odyhpc odyhpc  2292 Aug  6 23:40 build_packages.sh
-rw-rw-r--. 1 odyhpc odyhpc  3657 Aug  6 23:40 common.hpp
-rw-rw-r--. 1 odyhpc odyhpc   335 Aug  6 23:40 config_arch
-rw-rw-r--. 1 odyhpc odyhpc  7246 Aug  6 23:40 copybw.cpp
-rw-rw-r--. 1 odyhpc odyhpc 11764 Aug  6 23:40 gdrapi.c
-rw-rw-r--. 1 odyhpc odyhpc  4426 Aug  6 23:40 gdrapi.h
-rw-rw-r--. 1 odyhpc odyhpc  9688 Aug 22 20:47 gdrapi.o
-rw-rw-r--. 1 odyhpc odyhpc   284 Aug  6 23:40 gdrconfig.h
-rw-rw-r--. 1 odyhpc odyhpc  3496 Aug  6 23:40 gdrcopy.spec
drwxrwxr-x. 3 odyhpc odyhpc  4096 Aug 22 20:47 gdrdrv
drwxrwxr-x. 2 odyhpc odyhpc    21 Aug 22 20:45 init.d
-rw-rw-r--. 1 odyhpc odyhpc  1621 Aug  6 23:40 insmod.sh
lrwxrwxrwx. 1 odyhpc odyhpc    14 Aug 22 20:47 libgdrapi.so -> libgdrapi.so.1
lrwxrwxrwx. 1 odyhpc odyhpc    16 Aug 22 20:47 libgdrapi.so.1 -> libgdrapi.so.1.2
-rwxrwxr-x. 1 odyhpc odyhpc 14296 Aug 22 20:47 libgdrapi.so.1.2
-rw-rw-r--. 1 odyhpc odyhpc  1086 Aug  6 23:40 LICENSE
-rw-rw-r--. 1 odyhpc odyhpc  2281 Aug  6 23:40 Makefile
-rw-rw-r--. 1 odyhpc odyhpc  7614 Aug  6 23:40 memcpy_avx.c
-rw-rw-r--. 1 odyhpc odyhpc  5721 Aug  6 23:40 memcpy_sse41.c
-rw-rw-r--. 1 odyhpc odyhpc  6893 Aug  6 23:40 memcpy_sse.c
-rw-rw-r--. 1 odyhpc odyhpc  7005 Aug  6 23:40 README.md
-rw-rw-r--. 1 odyhpc odyhpc  4990 Aug  6 23:40 validate.cpp

```
Thanks.

pakmarkthub, 2019-08-22:
@afernandezody 
config_arch is supposed to have executable flag. Please try `chmod +x config_arg` then run make again.

afernandezody, 2019-08-22:
Pak,
Thanks. It went through (not sure why the permissions were wrong - insmod.sh had the same problem).

closed 2019-08-22 by afernandezody