NVIDIA/gdrcopy #78: Error when building dkms deb package

marsaev, 2019-08-22:
I'm seeing following error when building deb dkms package:

```bash
> ./build-deb-packages.sh
...
> dpkg-shlibdeps: error: no dependency information found for /usr/lib/x86_64-linux-gnu/libcuda.so.1 (used by debian/gdrcopy/usr/bin/sanity)
```

Which, i suppose, due to installing driver from downloaded *.run package. 

Error can be suppressed by adding rule:

```make
override_dh_shlibdeps:
        dh_shlibdeps --dpkg-shlibdeps-params=--ignore-missing-info
```
to `packages/debian/rules`, but i'm not sure whether this is the right way to maintain this.

marsaev, 2019-08-22:
Another issue in same command - if CUDA installed through runfile, not debian package, then this error happens:
```
dpkg-shlibdeps: error: cannot find library libcudart.so.10.1 needed by debian/gdrcopy/usr/bin/sanity (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
```
which is resolved by adding another flag to `dh_shlibdeps` in `override_dh_shlibdeps`: `-l${CUDA}/lib64`

pakmarkthub, 2019-08-22:
Thank you for reporting. I will investigate and fix this.

pakmarkthub, 2019-08-22:
@marsaev The fix has been merged into the master branch. Can I ask you to verify?

marsaev, 2019-08-23:
Hey Pak, works good for me! I'm also fine with 'validate' and 'copybw' not being part of gdrcopy package.

pakmarkthub, 2019-08-23:
Hi @marsaev. Nice to hear that it works. Actually, "validate" and "basic" have been merged into "sanity". We then distribute "copybw" and "sanity" with the package instead.

I am closing this issue because the problem has been fixed.

closed 2019-08-23 by pakmarkthub