NVIDIA/gdrcopy #92: failed to run the tests

wangzhezhe, 2019-09-22:
Hello, I built the tests successfully, but it failed for all the testing, maybe there is some misconfiguration for my building. Are there any suggestions for finding out the reasons?

Thanks a lot for your help!

```
zw241@gpu001:~/cworkspace/src/gdrcopy/tests$ ./copybw 
Assertion "cudaSuccess == result" failed at copybw.cpp:108
zw241@gpu001:~/cworkspace/src/gdrcopy/tests$ ./sanity 
Running suite(s): Sanity
Assertion "cudaSuccess == result" failed at sanity.cpp:178
Assertion "cudaSuccess == result" failed at sanity.cpp:220
Assertion "cudaSuccess == result" failed at sanity.cpp:319
Assertion "cudaSuccess == result" failed at sanity.cpp:447
Assertion "cudaSuccess == result" failed at sanity.cpp:527
Assertion "cudaSuccess == result" failed at sanity.cpp:607
Assertion "cudaSuccess == result" failed at sanity.cpp:756
Assertion "cudaSuccess == result" failed at sanity.cpp:868
Assertion "cudaSuccess == result" failed at sanity.cpp:1008
Assertion "cudaSuccess == result" failed at sanity.cpp:1117
Assertion "cudaSuccess == result" failed at sanity.cpp:1222
Assertion "cudaSuccess == result" failed at sanity.cpp:1337
0%: Checks: 12, Failures: 0, Errors: 12
sanity.cpp:170:E:Basic:basic:0: (after this point) Early exit with return value 1
sanity.cpp:212:E:Basic:basic_unaligned_mapping:0: (after this point) Early exit with return value 1
sanity.cpp:311:E:Data Validation:data_validation:0: (after this point) Early exit with return value 1
sanity.cpp:425:E:Invalidation:invalidation_access_after_gdr_close:0: (after this point) Early exit with return value 1
sanity.cpp:505:E:Invalidation:invalidation_access_after_cumemfree:0: (after this point) Early exit with return value 1
sanity.cpp:591:E:Invalidation:invalidation_two_mappings:0: (after this point) Early exit with return value 1
sanity.cpp:713:E:Invalidation:invalidation_fork_access_after_cumemfree:0: (after this point) Early exit with return value 1
sanity.cpp:861:E:Invalidation:invalidation_fork_after_gdr_map:0: (after this point) Early exit with return value 1
sanity.cpp:996:E:Invalidation:invalidation_fork_child_gdr_map_parent:0: (after this point) Early exit with return value 1
sanity.cpp:1091:E:Invalidation:invalidation_fork_map_and_free:0: (after this point) Early exit with return value 1
sanity.cpp:1216:E:Invalidation:invalidation_unix_sock_shared_fd_gdr_pin_buffer:0: (after this point) Early exit with return value 1
sanity.cpp:1315:E:Invalidation:invalidation_unix_sock_shared_fd_gdr_map:0: (after this point) Early exit with return value 1
```
The information of the device
```
zw241@gpu001:~/cworkspace/src/gdrcopy/tests$ nvidia-smi 
Sun Sep 22 17:43:13 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 390.116                Driver Version: 390.116                   |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Tesla M2070         On   | 00000000:09:00.0 Off |                    0 |
| N/A   N/A    P0    N/A /  N/A |    185MiB /  5301MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  Tesla M2070         On   | 00000000:0A:00.0 Off |                    0 |
| N/A   N/A   P12    N/A /  N/A |    137MiB /  5301MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   2  Tesla M2070         On   | 00000000:0D:00.0 Off |                    0 |
| N/A   N/A   P12    N/A /  N/A |    137MiB /  5301MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
...
```

pakmarkthub, 2019-09-23:
@wangzhezhe 

Hi,

Can I ask you to try `deviceQuery` or `bandwidthTest`? These two applications are included in the samples folder of your cuda package.

wangzhezhe, 2019-09-23:
Hello, @pakmarkthub, thanks for your reply. I use the cluster from my university, and they did not build them in the bin directory. It seems I do not have permission to build these two tools:

```
zw241@gpu003:/opt/sw/packages/cuda/10.0/samples/1_Utilities/deviceQuery$ module load gcc/5.4
zw241@gpu003:/opt/sw/packages/cuda/10.0/samples/1_Utilities/deviceQuery$ make
"/opt/sw/packages/cuda/10.0"/bin/nvcc -ccbin g++ -I../../common/inc  -m64    -gencode arch=compute_30,code=sm_30 -gencode arch=compute_35,code=sm_35 -gencode arch=compute_37,code=sm_37 -gencode arch=compute_50,code=sm_50 -gencode arch=compute_52,code=sm_52 -gencode arch=compute_60,code=sm_60 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_75,code=compute_75 -o deviceQuery.o -c deviceQuery.cpp
Assembler messages:
Fatal error: can't create deviceQuery.o: Permission denied
make: *** [deviceQuery.o] Error 1
zw241@gpu003:/opt/sw/packages/cuda/10.0/samples/1_Utilities/deviceQuery$ cd ..
zw241@gpu003:/opt/sw/packages/cuda/10.0/samples/1_Utilities$ s
bash: s: command not found
zw241@gpu003:/opt/sw/packages/cuda/10.0/samples/1_Utilities$ ls
UnifiedMemoryPerf  bandwidthTest  deviceQuery  deviceQueryDrv  p2pBandwidthLatencyTest  topologyQuery
zw241@gpu003:/opt/sw/packages/cuda/10.0/samples/1_Utilities$ cd bandwidthTest/
zw241@gpu003:/opt/sw/packages/cuda/10.0/samples/1_Utilities/bandwidthTest$ ls
Makefile  NsightEclipse.xml  bandwidthTest.cu  readme.txt
zw241@gpu003:/opt/sw/packages/cuda/10.0/samples/1_Utilities/bandwidthTest$ make
"/opt/sw/packages/cuda/10.0"/bin/nvcc -ccbin g++ -I../../common/inc  -m64    -gencode arch=compute_30,code=sm_30 -gencode arch=compute_35,code=sm_35 -gencode arch=compute_37,code=sm_37 -gencode arch=compute_50,code=sm_50 -gencode arch=compute_52,code=sm_52 -gencode arch=compute_60,code=sm_60 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_75,code=compute_75 -o bandwidthTest.o -c bandwidthTest.cu
Assembler messages:
Fatal error: can't create bandwidthTest.o: Permission denied
make: *** [bandwidthTest.o] Error 1
zw241@gpu003:/opt/sw/packages/cuda/10.0/samples/1_Utilities/bandwidthTest$ 
```

pakmarkthub, 2019-09-23:
@wangzhezhe You can copy them into your own directory and then compile the programs. I suspect that CUDA is not setup properly (might be an issue in your user environment). So, I would like you to verify with those standard applications first.

wangzhezhe, 2019-09-23:
Hello, @pakmarkthub thanks for your suggestion, I got following results on node with the tesla M2070:

```
zw241@gpu003:~/cworkspace/src/samples/1_Utilities/bandwidthTest$ nvidia-smi 
Mon Sep 23 16:13:39 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 390.116                Driver Version: 390.116                   |
|-------------------------------+----------------------+----------------------+
...
| N/A   N/A   P12    N/A /  N/A |    137MiB /  5301MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   7  Tesla M2070         On   | 00000000:33:00.0 Off |                    0 |
| N/A   N/A   P12    N/A /  N/A |    137MiB /  5301MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+
zw241@gpu003:~/cworkspace/src/samples/1_Utilities/bandwidthTest$./bandwidthTest 
[CUDA Bandwidth Test] - Starting...
Running on...

cudaGetDeviceProperties returned 35
-> CUDA driver version is insufficient for CUDA runtime version
CUDA error at bandwidthTest.cu:242 code=35(cudaErrorInsufficientDriver) "cudaSetDevice(currentDevice)" 
zw241@gpu003:~/cworkspace/src/samples/1_Utilities/bandwidthTest$ cd ../
UnifiedMemoryPerf/       bandwidthTest/           deviceQuery/             deviceQueryDrv/          p2pBandwidthLatencyTest/ topologyQuery/
zw241@gpu003:~/cworkspace/src/samples/1_Utilities/bandwidthTest$ cd ../deviceQuery
zw241@gpu003:~/cworkspace/src/samples/1_Utilities/deviceQuery$ ./deviceQuery 
./deviceQuery Starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

cudaGetDeviceCount returned 35
-> CUDA driver version is insufficient for CUDA runtime version
Result = FAIL
```

But on the node with the GeForce RTX 208... I got the following results:

```
zw241@gpu008:~/cworkspace/src/samples/1_Utilities/bandwidthTest$ nvidia-smi 
Mon Sep 23 16:00:11 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 430.26       Driver Version: 430.26       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce RTX 208...  On   | 00000000:1B:00.0 Off |                  N/A |
| 43%   81C    P2   261W / 250W |   8162MiB / 11019MiB |     95%      Default |
+-------------------------------+----------------------+----------------------+
|   1  GeForce RTX 208...  On   | 00000000:1C:00.0 Off |                  N/A |
| 45%   83C    P2   244W / 250W |   8162MiB / 11019MiB |     97%      Default |
+-------------------------------+----------------------+----------------------+
|   2  GeForce RTX 208...  On   | 00000000:1D:00.0 Off |                  N/A |
| 46%   84C    P2   241W / 250W |   8166MiB / 11019MiB |     97%      Default |
...
zw241@gpu008:~/cworkspace/src/samples/1_Utilities/bandwidthTest$ ls
Makefile  NsightEclipse.xml  bandwidthTest  bandwidthTest.cu  bandwidthTest.o  readme.txt
zw241@gpu008:~/cworkspace/src/samples/1_Utilities/bandwidthTest$ cd ..
zw241@gpu008:~/cworkspace/src/samples/1_Utilities$ ls
UnifiedMemoryPerf  bandwidthTest  deviceQuery  deviceQueryDrv  p2pBandwidthLatencyTest  topologyQuery
zw241@gpu008:~/cworkspace/src/samples/1_Utilities$ cd deviceQuery
zw241@gpu008:~/cworkspace/src/samples/1_Utilities/deviceQuery$ ls
Makefile  NsightEclipse.xml  deviceQuery  deviceQuery.cpp  deviceQuery.o  readme.txt
zw241@gpu008:~/cworkspace/src/samples/1_Utilities/deviceQuery$ ./deviceQuery 
[... 21 lines cut ...]
  Warp size:                                     32
  Maximum number of threads per multiprocessor:  1024
  Maximum number of threads per block:           1024
  Max dimension size of a thread block (x,y,z): (1024, 1024, 64)
  Max dimension size of a grid size    (x,y,z): (2147483647, 65535, 65535)
  Maximum memory pitch:                          2147483647 bytes
  Texture alignment:                             512 bytes
  Concurrent copy and kernel execution:          Yes with 3 copy engine(s)
  Run time limit on kernels:                     No
  Integrated GPU sharing Host Memory:            No
  Support host page-locked memory mapping:       Yes
  Alignment requirement for Surfaces:            Yes
  Device has ECC support:                        Disabled
  Device supports Unified Addressing (UVA):      Yes
  Device supports Compute Preemption:            Yes
  Supports Cooperative Kernel Launch:            Yes
  Supports MultiDevice Co-op Kernel Launch:      Yes
  Device PCI Domain ID / Bus ID / location ID:   0 / 140 / 0
  Compute Mode:
     < Default (multiple host threads can use ::cudaSetDevice() with device simultaneously) >

deviceQuery, CUDA Driver = CUDART, CUDA Driver Version = 10.2, CUDA Runtime Version = 10.0, NumDevs = 1
Result = PASS
```
For both nodes, the sanity test program fail.

wangzhezhe, 2019-09-23:
Hello, @pakmarkthub I reset the cuda version to make it match with the driver version , it works this time but there are some new problems:

```
zw241@gpu003:~/cworkspace/src/samples/1_Utilities/deviceQuery$ ./deviceQuery 
./deviceQuery Starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

Detected 1 CUDA Capable device(s)

Device 0: "Tesla M2070"
  CUDA Driver Version / Runtime Version          9.1 / 9.0
  CUDA Capability Major/Minor version number:    2.0
  Total amount of global memory:                 5302 MBytes (5559156736 bytes)
MapSMtoCores for SM 2.0 is undefined.  Default to use 64 Cores/SM
MapSMtoCores for SM 2.0 is undefined.  Default to use 64 Cores/SM
  (14) Multiprocessors, ( 64) CUDA Cores/MP:     896 CUDA Cores
  GPU Max Clock rate:                            1147 MHz (1.15 GHz)
  Memory Clock rate:                             1566 Mhz
  Memory Bus Width:                              384-bit
  L2 Cache Size:                                 786432 bytes
  Maximum Texture Dimension Size (x,y,z)         1D=(65536), 2D=(65536, 65535), 3D=(2048, 2048, 2048)
  Maximum Layered 1D Texture Size, (num) layers  1D=(16384), 2048 layers
  Maximum Layered 2D Texture Size, (num) layers  2D=(16384, 16384), 2048 layers
  Total amount of constant memory:               65536 bytes
  Total amount of shared memory per block:       49152 bytes
  Total number of registers available per block: 32768
  Warp size:                                     32
  Maximum number of threads per multiprocessor:  1536
  Maximum number of threads per block:           1024
  Max dimension size of a thread block (x,y,z): (1024, 1024, 64)
  Max dimension size of a grid size    (x,y,z): (65535, 65535, 65535)
  Maximum memory pitch:                          2147483647 bytes
  Texture alignment:                             512 bytes
  Concurrent copy and kernel execution:          Yes with 2 copy engine(s)
  Run time limit on kernels:                     No
  Integrated GPU sharing Host Memory:            No
  Support host page-locked memory mapping:       Yes
  Alignment requirement for Surfaces:            Yes
  Device has ECC support:                        Enabled
  Device supports Unified Addressing (UVA):      Yes
  Supports Cooperative Kernel Launch:            No
  Supports MultiDevice Co-op Kernel Launch:      No
  Device PCI Domain ID / Bus ID / location ID:   0 / 9 / 0
  Compute Mode:
     < Default (multiple host threads can use ::cudaSetDevice() with device simultaneously) >

deviceQuery, CUDA Driver = CUDART, CUDA Driver Version = 9.1, CUDA Runtime Version = 9.0, NumDevs = 1
Result = PASS

zw241@gpu003:~/cworkspace/src/samples/1_Utilities/bandwidthTest$ ./bandwidthTest 
[CUDA Bandwidth Test] - Starting...
Running on...

 Device 0: Tesla M2070
 Quick Mode

 Host to Device Bandwidth, 1 Device(s)
 PINNED Memory Transfers
   Transfer Size (Bytes)	Bandwidth(MB/s)
   33554432			6483.9

 Device to Host Bandwidth, 1 Device(s)
 PINNED Memory Transfers
   Transfer Size (Bytes)	Bandwidth(MB/s)
   33554432			6454.0

 Device to Device Bandwidth, 1 Device(s)
 PINNED Memory Transfers
   Transfer Size (Bytes)	Bandwidth(MB/s)
   33554432			105231.0

Result = PASS

NOTE: The CUDA Samples are not meant for performance measurements. Results may vary when GPU Boost is enabled.
zw241@gpu003:~/cworkspace/src/samples/1_Utilities/bandwidthTest$
zw241@gpu003:~/cworkspace/src/gdrcopy/build/bin$ ./sanity 
Running suite(s): Sanity
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:188
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:252
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:337
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:454
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:534
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:617
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:763
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:875
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1015
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1124
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1254
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1378
0%: Checks: 12, Failures: 1, Errors: 11
sanity.cpp:170:E:Basic:basic:0: (after this point) Early exit with return value 1
sanity.cpp:212:E:Basic:basic_unaligned_mapping:0: (after this point) Early exit with return value 1
sanity.cpp:311:E:Data Validation:data_validation:0: (after this point) Early exit with return value 1
sanity.cpp:425:E:Invalidation:invalidation_access_after_gdr_close:0: (after this point) Early exit with return value 1
sanity.cpp:505:E:Invalidation:invalidation_access_after_cumemfree:0: (after this point) Early exit with return value 1
sanity.cpp:591:E:Invalidation:invalidation_two_mappings:0: (after this point) Early exit with return value 1
sanity.cpp:730:F:Invalidation:invalidation_fork_access_after_cumemfree:0: Assertion 'read(read_fd, &cont, sizeof(int)) == sizeof(int)' failed: read(read_fd, &cont, sizeof(int)) == 0, sizeof(int) == 4
sanity.cpp:861:E:Invalidation:invalidation_fork_after_gdr_map:0: (after this point) Early exit with return value 1
sanity.cpp:996:E:Invalidation:invalidation_fork_child_gdr_map_parent:0: (after this point) Early exit with return value 1
sanity.cpp:1091:E:Invalidation:invalidation_fork_map_and_free:0: (after this point) Early exit with return value 1
sanity.cpp:1216:E:Invalidation:invalidation_unix_sock_shared_fd_gdr_pin_buffer:0: (after this point) Early exit with return value 1
sanity.cpp:1315:E:Invalidation:invalidation_unix_sock_shared_fd_gdr_map:0: (after this point) Early exit with return value 1
zw241@gpu003:~/cworkspace/src/gdrcopy/build/bin$ ./copybw 
GPU id:0 name:Tesla M2070 PCI domain: 0 bus: 9 device: 0
selecting device 0
testing size: 131072
rounded size: 131072
device ptr: 1302f00000
Assertion "!((g) == ((void*)0))" failed at copybw.cpp:137
```

pakmarkthub, 2019-09-23:
Hi @wangzhezhe,

This library supports only Tesla and Quadro GPUs. So, it would not work with your GeForce RTX card.

About the issue on your Tesla M2070 machine, have you already installed `gdrdrv`? This is a driver, which comes with this repository. If you have root privilege, you can simply run `sudo ./insmod.sh`. A better way is to build dep or rpm packages with the scripts we provide. Then install those packages.

wangzhezhe, 2019-09-23:
Hello, @pakmarkthub , thanks for your explanation. Is root privilege necessary to install this? I could ask the administrator team of the cluster to see if they could provide help.

pakmarkthub, 2019-09-23:
@wangzhezhe Unfortunately, yes. Please try copybw and sanity again after you get the driver installed.

closed 2019-09-23 by wangzhezhe