NVIDIA/gdrcopy #98: stdio.h should be included in include/gdrapi.h

azrael417, 2019-11-26:
it seems that stdio.h is not included in the header mentioned above which can yield error messages such as:

```common.cpp: In function ‘void gdrcopy::test::print_dbg(const char*, ...)’:
common.cpp:40:26: error: ‘stderr’ was not declared in this scope
                 vfprintf(stderr, fmt, ap);
                          ^
common.cpp:40:41: error: ‘vfprintf’ was not declared in this scope
                 vfprintf(stderr, fmt, ap);```

adding that header fixes it.

pakmarkthub, 2019-11-26:
Hi @azrael417,

Thank you for reporting this bug. The bug should have been fixed. Can I ask you to try and close this issue?

azrael417, 2019-11-27:
yes, thank you, looks good now.

closed 2019-11-27 by azrael417