NVIDIA/gdrcopy #103: Build error for sanity test

jw447, 2020-01-15:
I’m trying to build and install GDRCopy from source on RHEL when I had this following issue:

It seems error occurred when trying to build sanity test. Can someone please tell me how to resolve this? Looks like missing some gcc flags / gcc version here?

```
g++ -O2 -I /sw/summit/cuda/9.2.148/include -I ../include -I ../src -I /sw/summit/cuda/9.2.148/include -I /gpfs/alpine/proj-shared/csc143/jwang/local-build/check/check-0.13.0/install/include  -c -o sanity.o sanity.cpp
In file included from sanity.cpp:46:0:
common.hpp:98:9: error: 'testname' does not name a type
         testname = #__testname;                                                 \
         ^
sanity.cpp:174:1: note: in expansion of macro 'BEGIN_GDRCOPY_TEST'
 BEGIN_GDRCOPY_TEST(basic)
 ^
common.hpp:99:18: error: expected constructor, destructor, or type conversion before '(' token
         print_dbg("&&&& RUNNING " # __testname "\n");                           \
                  ^
sanity.cpp:174:1: note: in expansion of macro 'BEGIN_GDRCOPY_TEST'
 BEGIN_GDRCOPY_TEST(basic)
 ^
common.hpp:100:15: error: expected constructor, destructor, or type conversion before '(' token
         fflush(stdout);                                                         \
               ^
sanity.cpp:174:1: note: in expansion of macro 'BEGIN_GDRCOPY_TEST'
 BEGIN_GDRCOPY_TEST(basic)
 ^
common.hpp:101:15: error: expected constructor, destructor, or type conversion before '(' token
         fflush(stderr);                                                         \
               ^
sanity.cpp:174:1: note: in expansion of macro 'BEGIN_GDRCOPY_TEST'
 BEGIN_GDRCOPY_TEST(basic)
 ^
common.hpp:103:9: error: expected unqualified-id before 'if'
         if (__pid < 0) {                                                        \
         ^
sanity.cpp:174:1: note: in expansion of macro 'BEGIN_GDRCOPY_TEST'
 BEGIN_GDRCOPY_TEST(basic)
 ^
common.hpp:108:9: error: expected unqualified-id before 'if'
         if (__pid == 0)
         ^
sanity.cpp:174:1: note: in expansion of macro 'BEGIN_GDRCOPY_TEST'
 BEGIN_GDRCOPY_TEST(basic)
 ^
make[1]: *** [sanity.o] Error 1
make[1]: Leaving directory `/gpfs/alpine/csc143/proj-shared/jwang/local-build/gdrcopy/tests'
make: *** [exes] Error 2
```

System configuration:
Operating System: Red Hat Enterprise Linux Server 7.6 (Maipo)
CPU: IBM POWER9
GPU: NVIDIA Tesla v100
CUDA: 9.2.148
GCC: version 4.8.5
Check: 0.13.0
Subunit: 1.3.0

jw447, 2020-01-15:
Never mind. The issues went away after I switched to a stable version.

drossetti, 2020-01-22:
It is confirm so reopening

PR #105 (2020-01-22): Fixed compilation bug with libcheck-0.13
Problem:
- Issue #103.
- libcheck-0.13.0 changes the definition of START_TEST and END_TEST.
- sanity.cpp cannot be compiled because of syntax error due to this change.

This change:
- fixes #103.
- properly adds parentheses to BEGIN_GDRCOPY_TEST and END_GDRCOPY_TEST definitions.
- makes sanity.cpp compatible with libcheck-0.13.0 and prior versions.

Presubmit testings:
- locally on gc11 with libcheck-0.13 and 0.10.
closed 2020-01-22 by drossetti