NVIDIA/gdrcopy #109: check.h

ahban, 2020-02-25:
I got the following error and failed to find check.h. Where is it? Is it a requirement in other packages?

version: 2.0

```bash
cd tests && \
make CUDA=/home/aban/soft/cuda-8.0
make[1]: Entering directory '/home/aban/Downloads/gdrcopy-2.0/tests'
g++ -O2 -I /home/aban/soft/cuda-8.0/include -I ../include -I ../src -I /home/aban/soft/cuda-8.0/include  -c -o sanity.o sanity.cpp
sanity.cpp:38:19: fatal error: check.h: No such file or directory
compilation terminated.
```

likueimo, 2020-02-25:
I am also a grdcopy user, but I can help you deal with this issue.
You can read readme again , you will discover this information.

https://github.com/NVIDIA/gdrcopy

[image]

ahban, 2020-02-25:
> I am also a grdcopy user, but I can help you deal with this issue.
> You can read readme again , you will discover this information.
> 
> https://github.com/NVIDIA/gdrcopy
> 
> [image]

closed 2020-02-25 by ahban