NVIDIA/gdrcopy #113: Question about compatibility (2.0 -> 1.3)

likueimo, 2020-03-05:
I tried to run a [NGC lammps(24Oct2018)](https://ngc.nvidia.com/catalog/containers/hpc:lammps) singularity container.  
And it failed when I use more then 2 MPI process, showed some error message about gdr.  
I found that gdrcopy 1.3 inside container, and my host installed gdrcopy 2.0.

It worked with same container and same job last year, and then my cluster installed gdrcopy 1.3.

I guessed that the error may related about compatibility. 
Was gdrcopy 2.0 compatible gdrcopy 1.3 ?

```
[1583370992.161735] [gn0101:301910:0]    gdr_copy_md.c:113  UCX  ERROR gdr_pin_buffer failed. length :3735552 ret:25
[1583370992.161757] [gn0101:301910:0]         rcache.c:534  UCX  WARN  failed to register region 0x564d4dc29b00 [0x2b31b9a00000..0x2b31b9d90000]: Input/output error
[1583370992.161760] [gn0101:301910:0]         ucp_mm.c:105  UCX  ERROR failed to register address 0x2b31b9a00080 length 8168 on md[8]=gdr_copy: Input/output error
[1583370992.161763] [gn0101:301910:0]             dt.c:77   UCX  ERROR failed to register buffer with mem type domian
[1583370992.161769] [gn0101:301910:0]    gdr_copy_md.c:113  UCX  ERROR gdr_pin_buffer failed. length :3735552 ret:25
[1583370992.161774] [gn0101:301910:0]    gdr_copy_md.c:113  UCX  ERROR gdr_pin_buffer failed. length :3735552 ret:25                                         h mem type domian
```
[full output message](https://github.com/NVIDIA/gdrcopy/files/4290507/job_output.txt)

[slurm_script.txt](https://github.com/NVIDIA/gdrcopy/files/4290504/slurm_script.txt)

drossetti, 2020-03-06:
@likueimo I'm afraid that does not work.
You cannot mix gdrcopy 1.x user-space library with 2.x kernel-space driver. 
Feel free to close the issue if you don't have further questions.

likueimo, 2020-03-07:
@drossetti Thanks for your reply. 
If any possible, it will be nice to add some description about compatibility on README. 
Thanks!

closed 2020-03-07 by likueimo