NVIDIA/gdrcopy #114: sanity tests fail due to timeout

dmolka, 2020-03-05:
`sanity` test fail due to timeoouts:

``````

dmolka, 2020-03-05:
`copybw` runs without errors:

```
[root@cara-gpu-1 packages]# copybw
GPU id:0 name:Tesla V100-SXM2-32GB PCI domain: 0 bus: 97 device: 0
GPU id:1 name:Tesla V100-SXM2-32GB PCI domain: 0 bus: 98 device: 0
GPU id:2 name:Tesla V100-SXM2-32GB PCI domain: 0 bus: 137 device: 0
GPU id:3 name:Tesla V100-SXM2-32GB PCI domain: 0 bus: 138 device: 0
selecting device 0
testing size: 131072
rounded size: 131072
device ptr: 7f3a35a00000
DBG:  wc_mapping=1
map_d_ptr: 0x7f3a52b43000
info.va: 7f3a35a00000
info.mapped_size: 131072
info.page_size: 65536
info.mapped: 1
info.wc_mapping: 1
page offset: 0
user-space pointer:0x7f3a52b43000
writing test, size=131072 offset=0 num_iters=10000
DBG:  sse4_1=1 avx=1 sse=1 sse2=1
DBG:  using AVX implementation of gdr_copy_to_bar
write BW: 9669.12MB/s
reading test, size=131072 offset=0 num_iters=100
DBG:  using SSE4_1 implementation of gdr_copy_from_bar
read BW: 404.07MB/s
unmapping buffer
unpinning buffer
closing gdrdrv
```

drossetti, 2020-03-06:
Most probably timeouts are too short in the case when persistence mode is not enabled and therefore GPU context creation can take an unexpectedly long time.

drossetti, 2020-03-06:
can you try to run the test as below?
$ CUDA_VISIBLE_DEVICES=0 sanity

drossetti, 2020-03-10:
@dmolka one more thing, can you try the master branch? It should have an increased timeout.
If it still does not work, would you be able to experiment with increasing the timeout in https://github.com/NVIDIA/gdrcopy/blob/fb1982973212353dd2711c2bc5e79e97e6f73340/tests/sanity.cpp#L1568

drossetti, 2020-05-28:
@dmolka feel free to close the issue if the problem has been verified as fixed

dmolka, 2020-06-02:
> @dmolka feel free to close the issue if the problem has been verified as fixed

sorry, I did not get to this so far. I'll ask a colleague if he can do the additional tests.

rtschueter, 2020-06-03:
Tests with the current master run fine. It seems that the increased timeout fixed the issue.

drossetti, 2020-06-03:
> Tests with the current master run fine. It seems that the increased timeout fixed the issue.

thanks. 
closing this issue

closed 2020-06-03 by drossetti