NVIDIA/gdrcopy #115: gdrdrv.ko silently crashing

damianam, 2020-03-05:
`gdrdrv.ko` seems to be silently crashing in our system.

Initially, MPIs were failing when attempting to use `gdrcopy`. Reloading the driver made them work again, but it crashes again without obvious triggers or errors in the logs.

The sanity check also fails, which shows that the problem is unrelated to MPI. When a node is having problems with `gdrdrv`, we can't see an entry in `/proc/devices`. However, `lsmod` keeps showing the module as loaded.

The output for the sanity check is:

```
$ GDRCOPY_ENABLE_LOGGING=1 GDRCOPY_LOG_LEVEL=1 ./sanity -v
Running suite(s): Sanity
&&&& RUNNING basic
buffer size: 327680
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:181
&&&& FAILED basic
&&&& RUNNING basic_with_tokens
buffer size: 327680
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:218
&&&& FAILED basic_with_tokens
&&&& RUNNING basic_unaligned_mapping
First allocation: d_fa=0x2b50ffa00000, size=4
Second allocation: d_A=0x2b50ffa10200, size=65540, GPU-page-boundary 0x2b50ffa10000
d_A is unaligned
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:276
&&&& FAILED basic_unaligned_mapping
&&&& RUNNING data_validation
buffer size: 327680
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:358
&&&& FAILED data_validation
&&&& RUNNING invalidation_access_after_gdr_close
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:471
&&&& FAILED invalidation_access_after_gdr_close
&&&& RUNNING invalidation_access_after_cumemfree
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:547
&&&& FAILED invalidation_access_after_cumemfree
&&&& RUNNING invalidation_two_mappings
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:626
&&&& FAILED invalidation_two_mappings
&&&& RUNNING invalidation_fork_access_after_cumemfree
parent: Start
child: Start
child: waiting for cont signal from parent
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:772
Assertion "(read(read_fd, &cont, sizeof(int))) == (sizeof(int))" failed at sanity.cpp:740
&&&& FAILED invalidation_fork_access_after_cumemfree
&&&& RUNNING invalidation_fork_after_gdr_map
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:881
&&&& FAILED invalidation_fork_after_gdr_map
&&&& RUNNING invalidation_fork_child_gdr_map_parent
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1021
&&&& FAILED invalidation_fork_child_gdr_map_parent
&&&& RUNNING invalidation_fork_map_and_free
parent: Start
child: Start
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1134
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1134
&&&& FAILED invalidation_fork_map_and_free
&&&& RUNNING invalidation_unix_sock_shared_fd_gdr_pin_buffer
parent: Start
child: Start
child: Receiving fd from parent via unix socket
parent: Calling gdr_open
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1264
&&&& FAILED invalidation_unix_sock_shared_fd_gdr_pin_buffer
&&&& RUNNING invalidation_unix_sock_shared_fd_gdr_map
parent: Start
child: Start
child: Receiving fd from parent via unix socket
parent: Calling gdr_open
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1388
&&&& FAILED invalidation_unix_sock_shared_fd_gdr_map
&&&& RUNNING invalidation_fork_child_gdr_pin_parent_with_tokens
parent: Start
child: Start
ERR:  Error getting the gdrdrv driver version with ioctl error (errno=22). gdrdrv might be too old.
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1470
parent: CUDA generated tokens.p2pToken 0, tokens.vaSpaceToken 65024
&&&& FAILED invalidation_fork_child_gdr_pin_parent_with_tokens
100%: Checks: 14, Failures: 0, Errors: 0
```

Note that there might be a bug in the sanity check, where it reports 0 failures and errors, even though all the tests have failed.

The major device number reported in `dmesg` is the same as the device number of `/dev/gdrdrv`

We are in touch with some NVIDIA colleagues. They recommended opening a new issue here. At the moment it is not very clear how to proceed, since we can't reliably reproduce the problem once the driver has been reloaded. However, after waiting with the nodes online, `gdrdrv` ends up silently crashing again. The GPUs work fine though.

damianam, 2020-03-05:
BTW: our driver version is `418.87.00-1`, and we have `gdrcopy-2.0-3`

drossetti, 2020-03-05:
A kernel module crash dump should show up in the kernel log somehow.
Can you attach the output of 'dmesg -H', or simply 'dmesg' if -H is not supported on your system:
1) after a clean boot
2) after a crash event, if the machine is still alive
If it is not alive after the crash, can you set up a [netconsole](https://www.kernel.org/doc/Documentation/networking/netconsole.txt) to capture the crash dump?

damianam, 2020-03-06:
The machine is pretty much alive and even usable after a gdrdrv crash. I am not even sure if that can be qualified as crash. The system just acts as if the driver is not loaded, but `lsmod` lists it as loaded.

I have sent an email with the output of `dmesg -H` in 2 nodes (one freshly booted and healthy, and another one with `gdrdrv` unresponsive), and with the output of `nvidia-bug-report.sh`

drossetti, 2020-03-06:
The logs do not show anything interesting. As you said I exclude a crash of the kernel-mode driver component.

>> The system just acts as if the driver is not loaded, but lsmod lists it as loaded.

I suspect that for some reason, /dev/gdrdrv is either deleted or recreated with the wrong major number.

When the problem happens, can you double check that :
- lsmod shows gdrdrv
- /proc/devices shows gdrdrv with major number X, e.g.
$ grep gdrdrv /proc/devices
509 gdrdrv
- /dev/gdrdrv is a char inode with matching major X
[drossetti@pwr04 ~]$ ls -l /dev/gdrdrv 
crw-rw-rw-. 1 root root 509, 0 Feb 28 10:14 /dev/gdrdrv

drossetti, 2020-03-06:
BTW I'd love to see a debug tracing activated on your nodes.
That would print a kernel log in the gdrdrv_release() function to show that when the driver is  unloaded.
To do that, you need to pass an extra params when loading the kernel module:
$ /sbin/insmod src/gdrdrv/gdrdrv.ko dbg_enabled=1 info_enabled=1
- I'll need the kernel log to be captured and uploaded.
- please clean the kernel log before loading the driver with 'dmesg -c'
- capture the log after the error shows up with 'dmesg >/tmp/kernel.log'

damianam, 2020-03-09:
I mentioned it on the 1st post, but here it is summarized:

```
$ ./sanity
Running suite(s): Sanity
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:181
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:218
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:276
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:358
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:471
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:547
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:626
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:772
Assertion "(read(read_fd, &cont, sizeof(int))) == (sizeof(int))" failed at sanity.cpp:740
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:881
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1021
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1134
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1134
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1264
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1388
Assertion "!((g) == ((void*)0))" failed at sanity.cpp:1470
100%: Checks: 14, Failures: 0, Errors: 0
$ lsmod | grep gdr
gdrdrv                 17279  0
nvidia              18202459  94 nv_peer_mem,gdrdrv,nvidia_modeset,nvidia_uvm
$ grep gdr /proc/devices
$ ls -la /dev/gdrdrv
crw-rw-rw- 1 root root 226, 0 Mar  4 09:53 /dev/gdrdrv
$ dmesg -T | grep gdr
[Wed Mar  4 09:53:21 2020] gdrdrv:device registered with major number 226
[Wed Mar  4 09:53:21 2020] gdrdrv:dbg traces disabled, info traces disabled
```

I could enable debug tracing. The problem though, is that we can't reliably reproduce the problem. So I would have to enable the debug tracing everywhere and wait for it to crash. I am unsure of the overhead that could cause for users. For the time being, I have done that in a single node. I'll report back once the driver misbehaves again. If it takes too long, I'll enable debugging on other nodes to speed up the process.

drossetti, 2020-03-09:
I'm puzzled.
Is it really a bare-metal use-case, i.e. no containers, no virtualization stuff whatsoever ?
Is there some kind of background daemon which could possibly do something on our backs?

drossetti, 2020-03-09:
Can you create a reserved debug partition and submit the reproducer there?

drossetti, 2020-03-09:
Someone suggested that major==226 is used by nvidia_drm.
I don't see DRM being loaded there, but it could be somehow related.
Next time you reproduce, could you please attach the content of /proc/devices for more analysis ?

damianam, 2020-03-11:
There are no containers or VMs of any kind involved.

The node where I have loaded the module with `dbg_enabled=1 info_enabled=1` has not have the problem again yet. We do have 4 other nodes where the driver is loaded but not present in `/proc/devices`:

```
$ clush -bw jwc09n[030,033,108,141] "/sbin/lsmod | grep gdr" > gdrdrv_issues
---------------
jwc09n[030,033,108] (3)
---------------
gdrdrv                 17279  0 
nvidia              18202459  94 nv_peer_mem,gdrdrv,nvidia_modeset,nvidia_uvm
---------------
jwc09n141
---------------
gdrdrv                 17279  0 
nvidia              18202459  997 nv_peer_mem,gdrdrv,nvidia_modeset,nvidia_uvm

$ clush -bw jwc09n[030,033,108,141] "cat /proc/devices" >> gdrdrv_issues
---------------
jwc09n[108,141] (2)
---------------
Character devices:
  1 mem
  4 /dev/vc/0
  4 tty
  4 ttyS
  5 /dev/tty
  5 /dev/console
  5 /dev/ptmx
  7 vcs
 10 misc
 13 input
 29 fb
128 ptm
136 pts
162 raw
180 usb
188 ttyUSB
189 usb_device
195 nvidia-frontend
202 cpu/msr
203 cpu/cpuid
221 ss
222 kdump
223 trace
224 aux
225 mei
226 drm
227 nvidia-uvm
228 nvidia-nvswitch
229 nvidia-nvlink
230 infiniband_cm
231 infiniband_mad
231 infiniband_verbs
231 infiniband_cm
232 infiniband_mad
233 infiniband_verbs
234 sep4_1_u
235 sep4_1_b
236 sep4_1_s
237 sep4_1
238 socperf2_0
239 pax
240 rapl
241 ipmidev
242 ptp
243 pps
244 dimmctl
245 ndctl
246 hidraw
247 usbmon
248 bsg
249 hmm_device
250 watchdog
251 iio
252 rtc
253 dax
254 tpm

Block devices:
259 blkext
  9 md
254 mdp
---------------
jwc09n030
---------------
Character devices:
  1 mem
  4 /dev/vc/0
  4 tty
  4 ttyS
  5 /dev/tty
  5 /dev/console
  5 /dev/ptmx
  7 vcs
 10 misc
 13 input
 29 fb
128 ptm
136 pts
162 raw
180 usb
188 ttyUSB
189 usb_device
195 nvidia-frontend
202 cpu/msr
203 cpu/cpuid
221 ss
222 kdump
223 trace
224 aux
225 mei
226 drm
227 nvidia-uvm
228 nvidia-nvswitch
229 nvidia-nvlink
230 infiniband_cm
231 infiniband_mad
231 infiniband_verbs
231 infiniband_cm
232 infiniband_mad
233 sep4_1_u
234 sep4_1_b
235 sep4_1_s
236 sep4_1
237 socperf2_0
238 pax
239 infiniband_verbs
240 rapl
241 ipmidev
242 ptp
243 pps
244 dimmctl
245 ndctl
246 hidraw
247 usbmon
248 bsg
249 hmm_device
250 watchdog
251 iio
252 rtc
253 dax
254 tpm

Block devices:
259 blkext
  9 md
254 mdp
---------------
jwc09n033
---------------
Character devices:
  1 mem
  4 /dev/vc/0
  4 tty
  4 ttyS
  5 /dev/tty
  5 /dev/console
  5 /dev/ptmx
  7 vcs
 10 misc
 13 input
 29 fb
128 ptm
136 pts
162 raw
180 usb
188 ttyUSB
189 usb_device
195 nvidia-frontend
202 cpu/msr
203 cpu/cpuid
221 ss
222 kdump
223 trace
224 aux
225 mei
226 drm
227 nvidia-uvm
228 nvidia-nvswitch
229 nvidia-nvlink
230 infiniband_cm
231 infiniband_mad
231 infiniband_verbs
231 infiniband_cm
232 infiniband_mad
233 sep4_1_u
234 sep4_1_b
235 sep4_1_s
236 sep4_1
237 infiniband_verbs
238 socperf2_0
239 pax
240 rapl
241 ipmidev
242 ptp
243 pps
244 dimmctl
245 ndctl
246 hidraw
247 usbmon
248 bsg
249 hmm_device
250 watchdog
251 iio
252 rtc
253 dax
254 tpm

Block devices:
259 blkext
  9 md
254 mdp

$ clush -bw jwc09n[030,033,108,141] "ls -la /dev/gdrdrv" >> gdrdrv_issues
---------------
jwc09n030
---------------
crw-rw-rw- 1 root root 226, 0 Mar  3 09:19 /dev/gdrdrv
---------------
jwc09n033
---------------
crw-rw-rw- 1 root root 226, 0 Feb  7 14:05 /dev/gdrdrv
---------------
jwc09n108
---------------
crw-rw-rw- 1 root root 226, 0 Feb 21 11:23 /dev/gdrdrv
---------------
jwc09n141
---------------
crw-rw-rw- 1 root root 226, 0 Feb 20 08:57 /dev/gdrdrv

```

Unfortunately, these nodes do not have the debugging options enabled.

Regarding a possible daemon doing things behind our backs....... I guess it is possible, but doing what? Are you aware of this being a side effect of any particular operation/command/check?

Regarding creating a debug partition and submit a reproducer...... I could, but we don't have a reproducer. Once the driver is reloaded, we can't reliably make it break again.

Maybe our best chance is to "brute force" it and enable debugging in all the nodes, to maximize the chances to catch the issue.

drossetti, 2020-03-13:
@damianam the problem is clear... drm module has major 226, which is what gdrdrv was supposed to have been assigned, judging from /dev/gdrdrv.
It looks like when drm is loaded, it overrides gdrdrv.
I suspect the service script which is used to load gdrdrv.
Can you copy yours here ?
If different, can you switch to the one which comes with the RPM (packages/rhel/init.d/gdrcopy) ? That scripts makes sure that nvidia.ko is loaded before gdrdrv.ko is loaded. That could be good enough to change the module loading order and work around the problem.

drossetti, 2020-03-13:
in addition, in your service script, you could explicitly load the drm module before gdrdrv.

drossetti, 2020-03-13:
but, you know why is drm loaded in the first place?

drossetti, 2020-03-14:
looking at the kernel code https://elixir.bootlin.com/linux/v3.10.108/source/fs/char_dev.c#L95, it seems there can be cases where a dynamic-major character device driver might get assigned a major, then later a fixed-major driver can steal it.
In practice, the major reservation would not enforced for gdrdrv.

drossetti, 2020-03-18:
@damianam I'll go ahead and close this bug. Feel free to reopen it in case of further problems.

closed 2020-03-18 by drossetti

damianam, 2020-03-20:
Thanks for your help. We'll make sure the module is loaded at the end, and report back if we see the problem reoccurring.