NVIDIA/gdrcopy #131: dkms source not in deb (master)

NHDLarry, 2020-08-03:
Was just fooling with the master branch and found it isn't building the deb with sources in it. 

```
Preparing to unpack gdrdrv-dkms_2.0-2_amd64.deb ...
Unpacking gdrdrv-dkms:amd64 (2.0-2) over (2.0-2) ...
Setting up gdrdrv-dkms:amd64 (2.0-2) ...
Building for 4.15.0-112-generic
Building for architecture x86_64
This package appears to be a binaries-only package
 you will not be able to build against kernel 4.15.0-112-generic
 since the package source was not provided

```

drossetti, 2020-08-11:
It is confirmed and it is not a regression, e.g.

```
$ git checkout v2.0
...
$ CUDA=/usr/local/cuda-X.Y bash packages/build-deb-packages.sh
...
make[1]: Leaving directory '/tmp/gdr.EzKaQu/gdrdrv-dkms-2.0'
dh_installinit --name gdrdrv
dh_testdir
dh_testroot
dh_link
dh_strip
dh_compress
dh_fixperms
dh_installdeb
dh_shlibdeps
dh_gencontrol
dh_md5sums
dh_builddeb
dpkg-deb: building package 'gdrdrv-dkms' in '../gdrdrv-dkms_2.0_amd64.deb'.
 dpkg-genbuildinfo --build=binary
 dpkg-genchanges --build=binary >../gdrdrv-dkms_2.0_amd64.changes
dpkg-genchanges: info: binary-only upload (no source code included)
 dpkg-source --after-build gdrdrv-dkms-2.0
dpkg-buildpackage: info: binary-only upload (no source included)
```

drossetti, 2020-08-11:
branch fix_131 is an incomplete fix

drossetti, 2020-08-12:
> It is confirmed and it is not a regression, e.g.

my bad. v2.0 dkms package had the source embedded so it's a regression

commit 36e7db5f (2020-08-13): Merge pull request #133 from pakmarkthub/dev-issue-131

fixes #131 i.e. no-source bug in gdrdrv-dkms.deb
closed 2020-08-13 by drossetti