NVIDIA/gdrcopy #137: improve output of sanity test

drossetti, 2020-09-11:
customer feedback: there is a lot of noise in the output, even when not run with logs enabled, and the user may miss the last line which shows 0 errors.
```
Running suite(s): Sanity 
ERR:  error Invalid argument(22) while mapping handle ce00a0, rounded_size=69632 offset=0 
DBG:  wc_mapping=1 
DBG:  wc_mapping=1 
DBG:  wc_mapping=1 
DBG:  wc_mapping=1 
DBG:  wc_mapping=1 
DBG:  wc_mapping=1 
DBG:  wc_mapping=1 
DBG:  sse4_1=1 avx=1 sse=1 sse2=1 
DBG:  using AVX implementation of gdr_copy_to_bar 
DBG:  using SSE4_1 implementation of gdr_copy_from_bar 
warning: buffer size 327669 is not dword aligned, ignoring trailing bytes 
DBG:  wc_mapping=1 
DBG:  wc_mapping=1 
DBG:  wc_mapping=1 
DBG:  wc_mapping=1 
DBG:  wc_mapping=1 
CUDA error: CUDA_ERROR_INVALID_DEVICE 
Assertion "CUDA_SUCCESS == result" failed at sanity.cpp:67 
Assertion "(read(read_fd, &child_data, sizeof(int))) == (sizeof(int))" failed at sanity.cpp:817 
DBG:  wc_mapping=1 
ERR:  error Permission denied(13) while mapping handle ce0570, rounded_size=65536 offset=0 
CUDA error: CUDA_ERROR_INVALID_DEVICE 
Assertion "CUDA_SUCCESS == result" failed at sanity.cpp:67 
CUDA error: CUDA_ERROR_INVALID_DEVICE 
Assertion "CUDA_SUCCESS == result" failed at sanity.cpp:67 
CUDA error: CUDA_ERROR_INVALID_DEVICE 
Assertion "CUDA_SUCCESS == result" failed at sanity.cpp:67 
Assertion "sendfd(pair[1], fd) >= 0" failed at sanity.cpp:1399 
ERR:  ioctl error (errno=22) 
100%: Checks: 15, Failures: 0, Errors: 0 
```

pakmarkthub, 2020-12-09:
There actually are failures but the last line reports incorrectly. The PR should fix it.

pakmarkthub, 2021-01-04:
The PR has been merged. Closing.

closed 2021-01-04 by pakmarkthub