NVIDIA/gdrcopy #139: gdr_ping_buffer failed with -EINVAL during sanity.

umka-sev, 2020-09-16:
Hi developers,

I tries to play with gdrcopy and it looks failed on basic checks, can you help with it ?

Running suite(s): Sanity
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:188
CUDA error: CUDA_ERROR_INVALID_DEVICE
Assertion "CUDA_SUCCESS == result" failed at sanity.cpp:215
Assertion "(gdr_pin_buffer(g, d_A, A_size, 0, 0, &A_mh)) == (0)" failed at sanity.cpp:282
Assertion "(gdr_pin_buffer(pt->g, pt->d_buf, pt->size, 0, 0, &pt->mh)) == (0)" failed at sanity.cpp:1526
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:366
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:478
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:554
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh[i])) == (0)" failed at sanity.cpp:639
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:780
Assertion "(read(read_fd, &cont, sizeof(int))) == (sizeof(int))" failed at sanity.cpp:740
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:889
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:1029
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:1142
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:1142
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:1392
CUDA error: CUDA_ERROR_INVALID_DEVICE
Assertion "CUDA_SUCCESS == result" failed at sanity.cpp:1493
Assertion "(read(read_fd, &d_A, sizeof(CUdeviceptr))) == (sizeof(CUdeviceptr))" failed at sanity.cpp:1472
100%: Checks: 15, Failures: 0, Errors: 0

  410.261785] gdrdrv:gdrdrv_init:dbg traces disabled, info traces disabled
[  434.965975] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f600000 len=327680 p2p_token=0 va_space=0) failed [ret = -22]
[  435.826911] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f610000 len=66052 p2p_token=0 va_space=0) failed [ret = -22]
[  436.258205] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f6307600000 len=1048576 p2p_token=0 va_space=0) failed [ret = -22]
[  436.697484] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f600000 len=327680 p2p_token=0 va_space=0) failed [ret = -22]
[  437.138635] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f600000 len=65536 p2p_token=0 va_space=0) failed [ret = -22]
[  437.550297] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f600000 len=65536 p2p_token=0 va_space=0) failed [ret = -22]
[  437.975430] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f600000 len=65536 p2p_token=0 va_space=0) failed [ret = -22]
[  438.402851] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f600000 len=65536 p2p_token=0 va_space=0) failed [ret = -22]
[  438.848465] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f600000 len=65536 p2p_token=0 va_space=0) failed [ret = -22]
[  439.261069] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f6307600000 len=65536 p2p_token=0 va_space=0) failed [ret = -22]
[  439.909531] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f600000 len=65536 p2p_token=0 va_space=0) failed [ret = -22]
[  439.911090] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f600000 len=65536 p2p_token=0 va_space=0) failed [ret = -22]
[  441.619649] gdrdrv:gdrdrv_pin_buffer:nvidia_p2p_get_pages(va=7f630f600000 len=65536 p2p_token=0 va_space=0) failed [ret = -22]

# cat /proc/driver/nvidia/version
NVRM version: NVIDIA UNIX x86_64 Kernel Module  440.100  Fri May 29 08:45:51 UTC 2020
GCC version:  gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2)
# nvidia-smi
Wed Sep 16 20:10:08 2020
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.100      Driver Version: 440.100      CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 105...  Off  | 00000000:09:00.0  On |                  N/A |
|  0%   43C    P0    N/A / 120W |    450MiB /  4038MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      3197      G   /usr/lib/xorg/Xorg                           204MiB |
|    0      5682      G   /usr/bin/plasmashell                          44MiB |
|    0      5760      G   /usr/bin/konqueror                             1MiB |
|    0      6002      G   /usr/bin/akonadi_archivemail_agent             1MiB |
|    0      6004      G   /usr/bin/akonadi_imap_resource                 1MiB |
|    0      6005      G   /usr/bin/akonadi_imap_resource                 1MiB |
|    0      6009      G   /usr/bin/akonadi_mailfilter_agent              1MiB |
|    0      6026      G   /usr/bin/akonadi_sendlater_agent               1MiB |
|    0      6030      G   /usr/bin/akonadi_unifiedmailbox_agent          1MiB |
|    0      6190      G   /usr/bin/systemsettings5                      21MiB |
|    0      6479      G   ...-token=270C1652D5AA8CA88161ABCEB2BCB95A    56MiB |
|    0      6686      G   ...uest-channel-token=14720223284606212462     3MiB |
|    0      9053      G   ...AAAAAAAAAAAACAAAAAAAAAA= --shared-files   100MiB |
|    0     10731      G   /usr/lib/firefox/firefox                       1MiB |
+-----------------------------------------------------------------------------+

drossetti, 2020-09-18:
GDRCopy relies on GPUDirect RDMA which is only supported on Quadro and Tesla GPUs.

drossetti, 2020-09-30:
closing as not a bug

closed 2020-09-30 by drossetti