NVIDIA/gdrcopy #140: fakeroot is missing in deb package install instructions

jirikraus, 2020-09-28:
The deb package install instructions fail if fakeroot is not installed:

```console
gdrcopy-2.1/packages$ CUDA=/usr/local/cuda ./build-deb-packages.sh
Building debian package for the gdrcopy library ...
+ cd /data/jkraus/gdrcopy-2.1/packages

... [snip]

gdrcopy-2.1/src/memcpy_sse.c
+ cd /tmp/gdr.MbMC1p/gdrcopy-2.1
+ debuild --set-envvar=CUDA=/usr/local/cuda --set-envvar=PKG_CONFIG_PATH= -us -uc
 dpkg-buildpackage -rfakeroot -us -uc -ui
dpkg-buildpackage: error: fakeroot not found, either install the fakeroot
package, specify a command with the -r option, or run this as root
debuild: fatal error at line 1152:
dpkg-buildpackage -rfakeroot -us -uc -ui failed
Failed with error 29 to execute: debuild --set-envvar=CUDA=/usr/local/cuda --set-envvar=PKG_CONFIG_PATH= -us -uc
```

drossetti, 2020-09-30:
@pakmarkthub can you have a look?

pakmarkthub, 2020-09-30:
Confirm that fakeroot is needed.

closed 2020-11-04 by lukeyeager