NVIDIA/gdrcopy #144: Sanity Tests failing on Singularity Container

ming17, 2020-10-15:
I have installed GDRcopy version 2.1 on a Singularity Container but am having trouble verifying the installation with the provided sanity tests. Here are the outputs from running each test:

sanity Test:  
```
ming79@gpu050:~$ singularity exec --nv new.sif ~/gdrcopy/tests/sanity
/home/ming79/gdrcopy/tests/sanity: error while loading shared libraries: libsubunit.so.0: cannot open shared object file: No such file or directory
```

copybw Test:
```
singularity exec --nv new.sif ~/gdrcopy/tests/copybw
/home/ming79/gdrcopy/tests/copybw: symbol lookup error: /home/ming79/gdrcopy/tests/copybw: undefined symbol: cuDevicePrimaryCtxRelease_v2
```

copylat Test:
```
singularity exec --nv new.sif ~/gdrcopy/tests/copylat
GPU id:0; name: Tesla V100-SXM2-16GB; Bus id: 0000:89:00
GPU id:1; name: Tesla V100-SXM2-16GB; Bus id: 0000:8a:00
selecting device 0
device ptr: 0x7fb9fa000000
allocated size: 16777216

Assertion "!((g) == ((void*)0))" failed at copylat.cpp:183
```

I saw a similar issue raised and thus tested the CUDA Device Query and Bandwidth Tests, both of which passed.

The nvidia-smi execution results are as follows:
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.33.01    Driver Version: 440.33.01    CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Tesla V100-SXM2...  On   | 00000000:89:00.0 Off |                    0 |
| N/A   28C    P0    37W / 300W |      0MiB / 16160MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  Tesla V100-SXM2...  On   | 00000000:8A:00.0 Off |                    0 |
| N/A   31C    P0    38W / 300W |      0MiB / 16160MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+
```

Esthesia, 2020-11-26:
@ming17 Could you check that gdrdrv module inserted? If there is no gdrdrv module which can be inserted insmod.sh, that Assertion "!((g) == ((void*)0))" failed at copylat.cpp:183 occurs.

drossetti, 2021-01-30:
@ming17 it looks like either missing gdrdrv or missing /dev/gdrdrv inodes in container.
closing as not a bug
feel free to reopen

drossetti, 2021-01-30:
also see https://github.com/NVIDIA/gdrcopy/issues/138#issuecomment-695042447

closed 2021-01-30 by drossetti