NVIDIA/gdrcopy #145: DKMS for rpm

xkszltl, 2020-10-17:
I noticed only deb is built for dkms, while rpm is built as with kmod.
What's the reason for that?
Does this means on CentOS we have to rebuild gdrcopy every time there's a kernel update?
Can we enable DKMS for rpm as well?

drossetti, 2020-11-05:
We can look into that.
For reference:
- RHEL/CentOS utilizes kABI instead of DKMS, see https://access.redhat.com/discussions/3189552.
- DKMS is an option and is available on EPEL, see https://wiki.centos.org/HowTos/BuildingKernelModules.

xkszltl, 2020-11-08:
Does gdrcopy uses only stable ABI?

For DKMS, If I understand it correctly, nvidia GPU driver registers to dkms during installation (at least it asks for that).
So I think it should work.
And the main reason I bring up DKMS is because of that annoying GPU driver, which basically needs a rebuild every time after kernel update....

drossetti, 2021-01-14:
quick update:
- @pakmarkthub is working on #160 
- the interplay with GPU driver updates can be nasty:
  - ideally we would like to rebuild gdrdrv when a new GPU driver is installed, not necessarily associated to a kernel update. 
  - @pakmarkthub is there a way to mark such a dependency in dkms? I vaguely remember a mechanism in recent dkms versions.

pakmarkthub, 2021-01-20:
Hi @xkszltl,

PR #160 has been merged to the master branch. 
If you can test and give us some feedback, that would be very helpful.

xkszltl, 2021-01-21:
Thanks!
I don't have a non-prod machine currently that I can use to test this change.
However I noticed master branch doesn't respect PREFIX properly when `make install`.
Is this a bug?
[image]

pakmarkthub, 2021-01-29:
@xkszltl Sorry I missed your last message.

We use lower-case `prefix`.

```
$ make CUDA=/usr/local/cuda prefix=/tmp/gdrcopy install -j
cd src && \
make LIB_MAJOR_VER=2 LIB_MINOR_VER=1
make[1]: Entering directory '/home/pm/Projects/gdrcopy/src'
GDRAPI_ARCH=X86
make[1]: Leaving directory '/home/pm/Projects/gdrcopy/src'
cd tests && \
make CUDA=/usr/local/cuda
installing in /tmp/gdrcopy/lib /tmp/gdrcopy/include...
make[1]: Entering directory '/home/pm/Projects/gdrcopy/tests'
removed '/tmp/gdrcopy/lib/libgdrapi.so.2.1'
'src/libgdrapi.so.2.1' -> '/tmp/gdrcopy/lib/libgdrapi.so.2.1'
removed '/tmp/gdrcopy/include/gdrapi.h'
'include/gdrapi.h' -> '/tmp/gdrcopy/include/gdrapi.h'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/pm/Projects/gdrcopy/tests'
cd tests && make install DESTBIN=/tmp/gdrcopy/bin
make[1]: Entering directory '/home/pm/Projects/gdrcopy/tests'
installing exes in /tmp/gdrcopy/bin...
removed '/tmp/gdrcopy/bin/copybw'
'copybw' -> '/tmp/gdrcopy/bin/copybw'
removed '/tmp/gdrcopy/bin/copylat'
'copylat' -> '/tmp/gdrcopy/bin/copylat'
removed '/tmp/gdrcopy/bin/sanity'
'sanity' -> '/tmp/gdrcopy/bin/sanity'
make[1]: Leaving directory '/home/pm/Projects/gdrcopy/tests'
```

xkszltl, 2021-01-29:
@pakmarkthub 
Thought about that but there's a mix of `PREFIX`/`prefix` in this repo, is this intentional?
I decided to go with `PREFIX` at the end because it's more widely used and also mentioned in your doc:
[image]

drossetti, 2021-01-30:
@pakmarkthub if we settle on using prefix, let's update the README file

pakmarkthub, 2021-02-10:
@xkszltl Thank you for reporting this. README has been updated with lower-case `prefix`. Looking through old PRs, this was changed in #143.

drossetti, 2021-04-13:
closing as this has been implemented

closed 2021-04-13 by drossetti

cponder, 2022-06-04:
@xkszltl  
Have you tested GDRcopy on CentOS/Rocky/RedHat 8, or just CentOS 7?

xkszltl, 2022-06-05:
No, I don't have any CentOS 8 deployment given its death.

xkszltl, 2022-06-05:
> @xkszltl Thank you for reporting this. README has been updated with lower-case `prefix`. Looking through old PRs, this was changed in #143.

https://github.com/NVIDIA/gdrcopy/blob/6420035a89ba820942dbb9829af43ffdd9f1d8f8/packages/dkms/Makefile#L29

This one is still PREFIX.