NVIDIA/gdrcopy #148: DKMS builds without needing to install the driver

lukeyeager, 2020-11-05:
For containerized builds (which usually don't have the driver or kernel source installed inside the container), it would be nice to be able to build the deb dkms package without needing to mount `/usr/src/` and `/lib/modules/`. It seems like that should be possible since DKMS shouldn't need to do any actual "building" per-se until the package is installed. But it's possible I'm misunderstanding how DKMS works.

drossetti, 2020-11-05:
@pakmarkthub mind taking a look?

pakmarkthub, 2020-11-05:
Ok, I will have a look. Maybe we can simply set the package to source only and properly trigger `make` during installation.

pakmarkthub, 2020-12-11:
Closing this issue as the PR has been merged.

closed 2020-12-11 by pakmarkthub