NVIDIA/gdrcopy #152: What can I assume about thread safety?

bmerry, 2020-11-15:
Looking through the code I see comments like [https://github.com/NVIDIA/gdrcopy/blob/ba82488d137454228faf32ef593d0ed912c486be/src/gdrdrv/gdrdrv.c#L432](this), which make me concerned about thread safety. In the current version, what should I assume is thread-safe? In particular, is operating on *different* memory handles but with the same `gdr_t` safe?

drossetti, 2020-12-11:
> Looking through the code I see comments like this, which make me concerned about thread safety.
The problematic case is when one thread triggers an invalidation, e.g. cudaFree, while a second thread calls gdr_unpin_buffer.
It could theoretically happen, although it should be very very rare.

> In particular, is operating on different memory handles but with the same gdr_t safe?
Yes, it is safe.

closed 2020-12-11 by bmerry

nevion, 2021-05-06:
@drossetti  I do believe I've encountered this issue with concurrent (across threads) gdr_pin_buffer + gdr_maps and gdr_unmaps + gdr_unpin_buffer; completely different buffers in play... I've had situations where it is not rare.  
 It's interesting because ioctl returns error code -22 (unknown error), the addresses however are aligned and look good. User managing a lock is one way to deal with the problem and this seems to work, however I believe you are aiming for thread safety... the comment in that code and the nv driver lock look like a pinch of a situation; is there anything else you can do to make it safe?