NVIDIA/gdrcopy #154: 2.1 tag missing prefix "v"

bureddy, 2020-12-02:
not a priority issue. but it will be good to have a consistent pattern.  our sysadmin had to spend some time to figure it out why scripts are not working 

```
git tag -l
2.1
v1.3
v2.0
v2.0-beta.2
v2.0-beta.3
v2.0-beta.4
v2.1-beta1
v2.1-beta2
```

traveller42, 2020-12-09:
The link in the Changelog file uses v2.1, so it currently 404s.

drossetti, 2020-12-11:
@bureddy thanks for reporting the issue
@pakmarkthub please have a look

pakmarkthub, 2020-12-11:
Thank you @bureddy and @traveller42 for reporting this issue. I have fixed it. Now it should looks like this.

```
$ git tag -l
v1.3
v2.0
v2.0-beta.1
v2.0-beta.2
v2.0-beta.3
v2.0-beta.4
v2.1
v2.1-beta1
v2.1-beta2
```

Also the link in the Changelog file should work now. Can you confirm?

traveller42, 2020-12-11:
I can confirm that the link is working.

pakmarkthub, 2021-01-04:
Fixed. Closing.

closed 2021-01-04 by pakmarkthub