NVIDIA/gdrcopy #161: gdrdrv does on build on 5.10+

drossetti, 2021-01-15:
kzfee has been removed from the Linux kernel since 5.10-rc1
```
CC [M]  /home/lab/IMB/gdrcopy-2.1/src/gdrdrv/gdrdrv.o
/home/lab/IMB/gdrcopy-2.1/src/gdrdrv/gdrdrv.c: In function ‘gdrdrv_release’:
/home/lab/IMB/gdrcopy-2.1/src/gdrdrv/gdrdrv.c:384:9: error: implicit declaration of function ‘kzfree’; did you mean ‘kvfree’?
```

drossetti, 2021-01-15:
@pakmarkthub I think we can simply replace kzfree() with memset()+kfree(), right?

drossetti, 2021-01-15:
actually kzfree() has been renamed, see https://github.com/torvalds/linux/commit/453431a54934d917153c65211b2dabf45562ca88

drossetti, 2021-01-15:
probably simply using kfree() is enough... there is nothing really sensible in those mr objects

pakmarkthub, 2021-01-20:
The PR has been merged. Closing this issue.

closed 2021-01-20 by pakmarkthub