NVIDIA/gdrcopy #164: Successfully built but fail when testing

quan-maithanh, 2021-01-20:
Hi,

I am able to build and install the package using deb package without any errors. However when I run testing, it failed with sanity check (please see bellow). To check if the CUDA Toolkit is functional, I run some samples test in CUDA package and they work perfectly. Are there any suggests for my case? Appreciate a lot for your help! 

Below is output when I have the tests run after installing the package:

**quanmai@rhea:~/tools/gdrcopy$ sanity**
Running suite(s): Sanity
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:187
CUDA error: CUDA_ERROR_INVALID_DEVICE
Assertion "CUDA_SUCCESS == result" failed at sanity.cpp:214
Assertion "(gdr_pin_buffer(g, d_A, A_size, 0, 0, &A_mh)) == (0)" failed at sanity.cpp:279
Assertion "(gdr_pin_buffer(pt->g, pt->d_buf, pt->size, 0, 0, &pt->mh)) == (0)" failed at sanity.cpp:1512
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:362
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:473
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:548
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh[i])) == (0)" failed at sanity.cpp:632
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:772
Assertion "(read(read_fd, &cont, sizeof(int))) == (sizeof(int))" failed at sanity.cpp:733
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:880
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:1019
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:1131
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:1131
Assertion "(gdr_pin_buffer(g, d_ptr, size, 0, 0, &mh)) == (0)" failed at sanity.cpp:1379
CUDA error: CUDA_ERROR_INVALID_DEVICE
Assertion "CUDA_SUCCESS == result" failed at sanity.cpp:1479
Assertion "(read(read_fd, &d_A, sizeof(CUdeviceptr))) == (sizeof(CUdeviceptr))" failed at sanity.cpp:1458
6%: Checks: 15, Failures: 14, Errors: 0
sanity.cpp:196:F:Basic:basic:0: Failed
sanity.cpp:230:F:Basic:basic_with_tokens:0: Failed
sanity.cpp:330:F:Basic:basic_unaligned_mapping:0: Failed
sanity.cpp:1604:F:Basic:basic_child_thread_pins_buffer:0: Failed
sanity.cpp:430:F:Data Validation:data_validation:0: Failed
sanity.cpp:505:F:Invalidation:invalidation_access_after_gdr_close:0: Failed
sanity.cpp:584:F:Invalidation:invalidation_access_after_cumemfree:0: Failed
sanity.cpp:672:F:Invalidation:invalidation_two_mappings:0: Failed
sanity.cpp:833:F:Invalidation:invalidation_fork_access_after_cumemfree:0: Failed
sanity.cpp:978:F:Invalidation:invalidation_fork_after_gdr_map:0: Failed
sanity.cpp:1050:F:Invalidation:invalidation_fork_child_gdr_map_parent:0: Failed
sanity.cpp:1178:F:Invalidation:invalidation_fork_map_and_free:0: Failed
sanity.cpp:1402:F:Invalidation:invalidation_unix_sock_shared_fd_gdr_map:0: Failed
sanity.cpp:1495:F:Invalidation:invalidation_fork_child_gdr_pin_parent_with_tokens:0: Failed

**quanmai@rhea:~/tools/gdrcopy$ copybw**
GPU id:0; name: TITAN X (Pascal); Bus id: 0000:17:00
GPU id:1; name: Quadro P400; Bus id: 0000:b3:00
selecting device 0
testing size: 131072
rounded size: 131072
device ptr: 7fa892400000
closing gdrdrv

**quanmai@rhea:~/tools/gdrcopy$ copylat**
GPU id:0; name: TITAN X (Pascal); Bus id: 0000:17:00
GPU id:1; name: Quadro P400; Bus id: 0000:b3:00
selecting device 0
device ptr: 0x7f440e400000
allocated size: 16777216

closing gdrdrv

pakmarkthub, 2021-01-21:
Hi @quan-maithanh,

The GPU you used does not support GPUDirect RDMA. Please see https://github.com/NVIDIA/gdrcopy#requirements.

quan-maithanh, 2021-01-21:
Hi @pakmarkthub 

Thanks for your reply, appreciate it a lot! 

I did think that my TITAN X GPU has GPUDirect RDMA support as  its architecture (Pascal) is in the support list?

pakmarkthub, 2021-01-21:
You need to run it with a Quadro or Tesla class GPU. Can you try running it with your Quadro P400? You can set `export CUDA_VISIBLE_DEVICES=1` and then rerun the test.

quan-maithanh, 2021-01-21:
Yes, the test works perfectly with the Quadro one, but with the GTX it is unfortunately not supported yet. Thanks for your help! I am going to close this topic in a couple of hours.

closed 2021-01-21 by quan-maithanh