NVIDIA/gdrcopy #176: Detected unexpected system page size when running on aarch64

pakmarkthub, 2021-02-10:
Error in `gdr_open`.
```
+ GDRCOPY_ENABLE_LOGGING=1
+ GDRCOPY_LOG_LEVEL=1
+ ./copybw
GPU id:0; name: NVIDIA Tesla V100-PCIE-32GB; Bus id: 0000:01:00
GPU id:1; name: NVIDIA Tesla V100-PCIE-32GB; Bus id: 0000:81:00
selecting device 0
testing size: 131072
rounded size: 131072
device ptr: ffffb0600000
ERR:  detected unexpected system page size
gdr_open error: Is gdrdrv driver installed and loaded?
ERROR: while running copybw with exit code 1
```

On that node:
```
page_size from sysconf is 4096
PAGE_SIZE from define is 65536
```

On a different aarch64 node:
```
page_size from sysconf is 65536
PAGE_SIZE from define is 65536
```

drossetti, 2021-02-11:
which Arm server system is this?

drossetti, 2021-02-11:
possible WAR seems to use sysconf(_SC_PAGESIZE) in gdrapi.c.
I'd get rid of PAGE_* macros altogether.

closed 2021-02-25 by pakmarkthub