NVIDIA/gdrcopy #187: Building deb package builds the tests, which adds dependency on CUDA when building debs

cliffwoolley, 2021-04-05:
When running `build-deb-packages.sh`, it ends up implicitly building the tests (both because `debian/rules` builds the `exes` and, even if you turn that off, because the build process installs the built debs at the end, and `make install` also leads to `exes_install` which again builds the tests.

This is unfortunate, because building the actual deb packages does not actually depend on the CUDA Toolkit being installed; only the tests depend on CUDA.  So for example if I hack up the Makefile as follows, then `build-deb-packages.sh` will run fine without CUDA installed:

```
--- Makefile.orig       2021-04-04 23:17:02.799050000 -0700
+++ Makefile.new        2021-04-04 23:18:11.437725000 -0700
@@ -59,10 +59,8 @@
        $(MAKE) LIB_MAJOR_VER=$(LIB_MAJOR_VER) LIB_MINOR_VER=$(LIB_MINOR_VER)

 exes: lib
-       cd tests && \
-       $(MAKE) CUDA=$(CUDA)

-install: lib_install exes_install
+install: lib_install

 lib_install: lib
        @ echo "installing in $(DESTLIB) $(DESTINC)..." && \
```

drossetti, 2021-04-13:
@cliffwoolley would moving the tests to a separate package solve the problem?

cliffwoolley, 2021-04-14:
As far as I can tell, the tests aren't packaged anyway.  It seemed like they were getting built more as a side-effect.

pakmarkthub, 2021-04-19:
@cliffwoolley Sorry for the delay. I just have time to focus on this project.

I rechecked and the tests are in the deb package (see below, you can see copybw, copylat, sanity). Are they not in the package you generated using the script?

Let's go back to your request. Is my understanding correct that you don't want to install CUDA in order to use the script? I don't think you want the CUDA dependency be removed from `gdrcopy.deb` because there is no such dependency check in the first place.

If my understanding is correct, I think I can provide an option in the `build-deb-packages.sh` script to **NOT** build the tests. In that case, we also need to put the tests into another package as Davide suggested.

```
$ dpkg -c gdrcopy_2.2-1_amd64.ubuntu20_04.deb 
drwxr-xr-x root/root         0 2021-02-02 03:59 ./
drwxr-xr-x root/root         0 2021-02-02 03:59 ./usr/
drwxr-xr-x root/root         0 2021-02-02 03:59 ./usr/bin/
-rwxr-xr-x root/root     26792 2021-02-02 03:59 ./usr/bin/copybw
-rwxr-xr-x root/root     26792 2021-02-02 03:59 ./usr/bin/copylat
-rwxr-xr-x root/root     96352 2021-02-02 03:59 ./usr/bin/sanity
drwxr-xr-x root/root         0 2021-02-02 03:59 ./usr/include/
-rw-r--r-- root/root      5353 2021-02-02 03:59 ./usr/include/gdrapi.h
drwxr-xr-x root/root         0 2021-02-02 03:59 ./usr/lib/
drwxr-xr-x root/root         0 2021-02-02 03:59 ./usr/lib/x86_64-linux-gnu/
-rw-r--r-- root/root     22848 2021-02-02 03:59 ./usr/lib/x86_64-linux-gnu/libgdrapi.so.2.2
drwxr-xr-x root/root         0 2021-02-02 03:59 ./usr/share/
drwxr-xr-x root/root         0 2021-02-02 03:59 ./usr/share/doc/
drwxr-xr-x root/root         0 2021-02-02 03:59 ./usr/share/doc/gdrcopy/
-rw-r--r-- root/root      4209 2021-02-02 03:59 ./usr/share/doc/gdrcopy/README.Debian.gz
-rw-r--r-- root/root       839 2021-02-02 03:59 ./usr/share/doc/gdrcopy/changelog.Debian.gz
-rw-r--r-- root/root      1268 2021-02-02 03:59 ./usr/share/doc/gdrcopy/copyright
lrwxrwxrwx root/root         0 2021-02-02 03:59 ./usr/lib/x86_64-linux-gnu/libgdrapi.so -> libgdrapi.so.2
lrwxrwxrwx root/root         0 2021-02-02 03:59 ./usr/lib/x86_64-linux-gnu/libgdrapi.so.2 -> libgdrapi.so.2.2
```

cliffwoolley, 2021-04-19:
In that case, putting the tests into a different package would make a lot of sense, I agree.

For backward compatibility purposes for people who might have relied on the tests getting installed with `apt install gdrcopy`, maybe it should go like this:

* _gdrcopy_ -> this becomes a metapackage, depends on all three of the following:
* **libgdrapi** -> just the userspace libs
* **gdrcopy-tests** -> just the tests
* _gdrdrv_ -> the already separate kernel driver package, on which gdrcopy package already depends, per https://github.com/NVIDIA/gdrcopy/issues/188

pakmarkthub, 2021-07-08:
The PRs have been merged. Closing this issue.

closed 2021-07-08 by pakmarkthub