NVIDIA/gdrcopy #189: use new CUDA 11.3 device attributes

drossetti, 2021-04-08:
In CUDA 11.3, CU_DEVICE_ATTRIBUTE_GPU_DIRECT_RDMA_SUPPORTED has been added.
Tracking here changes to deploy the new attribute, for example to print a meaningful error message.

drossetti, 2021-04-13:
for example, one could use something like:
```
#if CUDART_VERSION >= 11030
cudaDriverGetVersion(&driverVersion);
if (driverVersion > 11030) {
  cudaDeviceGetAttribute(attr,
      cudaDevAttrGPUDirectRDMASupported, device);
  gdrdmaSupported = true;
}
#endif
```

drossetti, 2021-04-13:
note that at the moment, libgdrapi does not have a dependency on libcuda, and it is not clear whether adding such dependency to call `cudaDeviceGetAttribute` is a good idea.
A compromise could be 1) to add that check in the test apps and 2) to document that the library does not explicitly check for GDR support and that users can run test apps to get appropriate diagnostic.

pakmarkthub, 2021-07-08:
The PR has been merged. Closing this issue.

closed 2021-07-08 by pakmarkthub