NVIDIA/gdrcopy #191: error while loading shared libraries: libgdrapi.so.2: cannot open shared object file: No such file or directory

jiangxiaobin96, 2021-04-12:
I have some problems while installing gdrcopy.
```
root@iZ8vbcavfdwt3inq0sbsj0Z:~/gdrcopy# make prefix=/usr/local CUDA=/usr/local/cuda-10.2 all install
GDRAPI_ARCH=X86
cd src/gdrdrv && \
make 
make[1]: Entering directory '/root/gdrcopy/src/gdrdrv'
Picking NVIDIA driver sources from NVIDIA_SRC_DIR=/usr/src/nvidia-450-450.80.02/nvidia. If that does not meet your expectation, you might have a stale driver still around and that might cause problems.
make[2]: Entering directory '/usr/src/linux-headers-4.4.0-198-generic'
  Building modules, stage 2.
  MODPOST 2 modules
make[2]: Leaving directory '/usr/src/linux-headers-4.4.0-198-generic'
make[1]: Leaving directory '/root/gdrcopy/src/gdrdrv'
cd src && \
make LIB_MAJOR_VER=2 LIB_MINOR_VER=2
make[1]: Entering directory '/root/gdrcopy/src'
GDRAPI_ARCH=X86
make[1]: Leaving directory '/root/gdrcopy/src'
cd tests && \
make CUDA=/usr/local/cuda-10.2
make[1]: Entering directory '/root/gdrcopy/tests'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/root/gdrcopy/tests'
installing in /usr/local/lib /usr/local/include...
removed '/usr/local/lib/libgdrapi.so.2.2'
'src/libgdrapi.so.2.2' -> '/usr/local/lib/libgdrapi.so.2.2'
removed '/usr/local/include/gdrapi.h'
'include/gdrapi.h' -> '/usr/local/include/gdrapi.h'
cd tests && make install DESTBIN=/usr/local/bin
make[1]: Entering directory '/root/gdrcopy/tests'
installing exes in /usr/local/bin...
removed '/usr/local/bin/copybw'
'copybw' -> '/usr/local/bin/copybw'
removed '/usr/local/bin/copylat'
'copylat' -> '/usr/local/bin/copylat'
removed '/usr/local/bin/sanity'
'sanity' -> '/usr/local/bin/sanity'
make[1]: Leaving directory '/root/gdrcopy/tests'
root@iZ8vbcavfdwt3inq0sbsj0Z:~/gdrcopy# sudo ./insmod.sh 
INFO: driver major is 241 
INFO: creating /dev/gdrdrv inode
root@iZ8vbcavfdwt3inq0sbsj0Z:~/gdrcopy# cd tests/
root@iZ8vbcavfdwt3inq0sbsj0Z:~/gdrcopy/tests# ./sanity 
./sanity: error while loading shared libraries: libgdrapi.so.2: cannot open shared object file: No such file or directory
```

pakmarkthub, 2021-04-12:
Hi @jiangxiaobin96 ,

Have you checked your `$LD_LIBRARY_PATH`? Does that environment variable contain `/usr/local/lib`?

jiangxiaobin96, 2021-04-12:
Oh, thanks very much. The problem has been solved.

closed 2021-04-12 by jiangxiaobin96

smallllwhite, 2023-08-09:
I meet the same problem, how do you solve it?

freebacca, 2023-10-12:
You can try to specify the location of your library by setting the environment variable LD_LIBRARY_PATH.
For example:
`export LD_LIBRARY_PATH=/usr/local/lib`

pyt-hnu, 2023-11-23:
> You can try to specify the location of your library by setting the environment variable LD_LIBRARY_PATH. For example: `export LD_LIBRARY_PATH=/usr/local/lib`

I can't solve it by your method,  is there another way?

pakmarkthub, 2023-11-24:
Hi @pyt-hnu,
This is likely your linker issue. You may need to open a new shell, do `export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH`, and run your GDRCopy application again. If it does not help, you may need to do `ldd <application>`, check if there are other missing links, and figure out where those libraries are installed on your system.

pyt-hnu, 2023-11-30:
> Hi @pyt-hnu, This is likely your linker issue. You may need to open a new shell, do `export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH`, and run your GDRCopy application again. If it does not [...]

Below is my command. and I followed your instruct, It's so strange.
```
ta@ta:~/Workspace/pyc/gdrcopy-master/tests$ sudo ./sanity 
./sanity: error while loading shared libraries: libgdrapi.so.2: cannot open shared object file: No such file or directory
ta@ta:~/Workspace/pyc/gdrcopy-master/tests$ ldd sanity
        linux-vdso.so.1 (0x00007ffccb3cd000)
        libcuda.so.1 => /lib/x86_64-linux-gnu/libcuda.so.1 (0x00007fd25919c000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd259179000)
        libgdrapi.so.2 => /usr/local/lib/libgdrapi.so.2 (0x00007fd259172000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fd259168000)
        libsubunit.so.0 => /lib/x86_64-linux-gnu/libsubunit.so.0 (0x00007fd259163000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd258f81000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd258e30000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd258c3e000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd258c38000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fd25ae54000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd258c1d000)
ta@ta:~/Workspace/pyc/gdrcopy-master/tests$ 
```

pakmarkthub, 2023-12-01:
Hi @pyt-hnu

1. Why did you run `sudo ./sanity`? Generally, `sudo` is not required to run `gdrcopy_sanity`. 
2. You ran `ldd sanity` without `sudo`. Also, if you already have `sanity` installed, `ldd sanity` and `ldd ./sanity` may give different output. In addition, `ldd ./sanity` and `sudo ldd ./sanity` may also give different output because of different environment variables used with and without `sudo`. So, please use the exact same command and environment as when you run `sanity`.
3. Have you checked that `/usr/local/lib/libgdrapi.so.2` really exists? This file is likely a symlink. Is the link broken?
4. Because you run `sudo ./sanity`, does your `sudo` have read access to the actual `libgdrapi.so.xxx` file? Remind that `libgdrapi.so.2` is just a symlink. You need to follow the symlink all the way until you find the actual library file.
5. You seemed to run `./sanity`, which was compiled from the master branch. Is `/usr/local/lib/libgdrapi.so.2` the same version as the master branch? You may run into a missing symbol issue if `/usr/local/lib/libgdrapi.so.2` is older than the master branch.