NVIDIA/gdrcopy #205: Question about getting version info

sakaia, 2021-07-06:
In the following command, it cannot get the version info.

What is the correct way to get driver info?

```
lsmod | grep gdrdrv
modinfo gdrdrv
```
Typical output is follows

```
$ lsmod | grep gdrdrv
gdrdrv                 24576  0
nvidia              34037760  19 nvidia_uvm,nv_peer_mem,gdrdrv,nvidia_modeset
$ modinfo gdrdrv
modinfo: ERROR: Module gdrdrv not found.
```

pakmarkthub, 2021-07-06:
Hi @sakaia,

I believe that gdrdrv is not registered to your system. How did you install the gdrdrv driver? Did you use `make drv_install` or install from the deb or rpm package? 

Another method is to use `gdr_driver_get_version`. See https://github.com/NVIDIA/gdrcopy/blob/master/include/gdrapi.h#L138.

sakaia, 2021-07-07:
@pakmarkthub  Thank you for your suggestion. The problem solved by doing  `make drv_install`

Previous instruction is `make; insmod.sh` by following some documents. 
but it needs to follow upstream document.

sakaia, 2021-07-07:
The problem of lsmod/modinfo is solved.

closed 2021-07-07 by sakaia

pakmarkthub, 2021-07-07:
`make drv_install` also registers `gdrdrv` to the module management on your system but we don't provide a script to uninstall that.

If possible, I would like to recommend you to build and use deb or rpm packages instead. Scriptlets inside those packages will take care of dependency enforcement as well as cleaning when you uninstall them.