NVIDIA/gdrcopy #208: benchmark and eventually support MOVDIRI/MOVDIR64B on Sapphire Rapids Xeon

drossetti, 2021-07-16:
from Intel manuals:
```
Unlike WC stores and stores with non-temporal hint, direct-stores are eligible for immediate
eviction from the write-combining buffer, and thus not combined with younger stores (including direct-stores)
to the same address. Older WC and non-temporal stores held in the write-combing buffer may be combined with
younger direct stores to the same address. Direct stores are weakly ordered relative to other stores. Software that
desires stronger ordering should use a fencing instruction (MFENCE or SFENCE) before or after a direct store to
enforce the ordering desired.
```

imaginary-person, 2021-07-17:
As of now, Sapphire Rapid Xeon is only available to select corporate Intel customers.

drossetti, 2022-12-09:
@pakmarkthub this might be fun :)