NVIDIA/gdrcopy #211: Does GPUDirect RDMA support for 2080Ti?

ReyRen, 2021-08-19:
Hello, 

I'm really confused about "GPUDirect RDMA requires NVIDIA Tesla or Quadro class GPUs based on Kepler, Pascal, Volta, or Turing" in README. Does this means 2080Ti, which is the Pascal arch also enable GPUDirect RDMA? However,  refer to https://docs.nvidia.com/cuda/gpudirect-rdma/index.html says "GPUDirect RDMA is available on both Tesla and Quadro GPUs".

I'd appreciate with any help from you.

Thanks,
Rey

pakmarkthub, 2021-08-19:
Hi Rey,

I believe that you are referring to this product https://www.nvidia.com/en-us/geforce/graphics-cards/rtx-2080-ti/. This is a GeForce-class GPU, so it does not support GPUDirect RDMA. You would need a Tesla-class or a Quadro-class GPU such as P100, V100, etc.

ReyRen, 2021-08-19:
> I believe that you are referring to this product https://www.nvidia.com/en-us/geforce/graphics-cards/rtx-2080-ti/. This is a GeForce-class GPU, so it does not support GPUDirect RDMA. You would need a Tesla-class or a Quadro-class GPU such as P100, V100, etc.

@pakmarkthub  👍🏽 Thanks,  I'd close this.

closed 2021-08-19 by ReyRen