NVIDIA/gdrcopy #221: Question about GDRCopy bandwidths

tropuq, 2022-01-20:
Hi,

I have a question about the performance of GDRCopy. I noticed in README that the write BW that you could achieve with Tesla V100 was 9224.36 MB/s. I checked some other projects that use GPUDirect RDMA with third-party devices and they were able to achieve better performance (e.g. https://github.com/NVIDIA/l2fwd-nv#io-forwarding - 94 Gbps for packets of size 1024 which is around 11750 MB/s). What do third-party devices do differently that they can achieve better performance with GPUDirect RDMA?

I'd appreciate any explanations.

Thanks

drossetti, 2022-01-26:
On common platforms, GDRCopy relies on CPU-initiated PCIe transactions targeting the GPU PCIe BAR. The effective write BW is limited by the quality of the PCIe traffic generated by the CPU. Most probably the CPU generates 32B or 64B PCIe MemWr transactions, which a) are affected by protocol overhead, b) may be emitted at lower than peak rate by the CPU, among other things.

tropuq, 2022-02-10:
Ok, thank you. I'm closing the issue.

closed 2022-02-10 by tropuq