NVIDIA/gdrcopy #224: Failed run sanity test.

zhang3689, 2022-03-09:
envir:
centos 7.2
cuda: 11.0
gpu: A100-SXM4-40GB
when I run sanity test, meet the errors as pic:
[image]

But the copybw / copylat / apiperf tests  is ok!
[image]
[image]
[image]

Are there any suggestions for finding out the reasons?
Thanks a lot for your help!

pakmarkthub, 2022-03-09:
Hi @myanzhang ,

Can you run `sanity -v` and post the output?

myanzhang, 2022-03-09:
邮件已收到，谢谢

myanzhang, 2022-03-09:
[root@ts-6ab12923e4f84b41a1dec977fcf2a978-launcher ~/gdrcopy/tests]# ./sanity -v
Running suite(s): Sanity
&&&& RUNNING basic_cumemalloc
buffer size: 327680
&&&& PASSED basic_cumemalloc
&&&& RUNNING basic_with_tokens
buffer size: 327680
&&&& PASSED basic_with_tokens
&&&& RUNNING basic_unaligned_mapping
First allocation: d_fa=0x7f1263200000, size=4
Second allocation: d_A=0x7f1263220200, size=65540, GPU-page-boundary 0x7f1263220000
d_A is unaligned
Try mapping d_A as is.
Mapping d_A failed as expected.
Align d_A and try mapping it again.
Pin and map aligned address: d_aligned_A=0x7f1263230000, offset=65024, size=516
&&&& PASSED basic_unaligned_mapping
&&&& RUNNING basic_child_thread_pins_buffer_cumemalloc
spawning single child thread
pinning
Assertion "(gdr_pin_buffer(pt->g, pt->d_buf, pt->size, 0, 0, &pt->mh)) == (0)" failed at sanity.cpp:1751
&&&& FAILED basic_child_thread_pins_buffer_cumemalloc
&&&& RUNNING basic_vmmalloc
buffer size: 327680
&&&& PASSED basic_vmmalloc
&&&& RUNNING basic_child_thread_pins_buffer_vmmalloc
spawning single child thread
pinning
Assertion "(gdr_pin_buffer(pt->g, pt->d_buf, pt->size, 0, 0, &pt->mh)) == (0)" failed at sanity.cpp:1751
&&&& FAILED basic_child_thread_pins_buffer_vmmalloc
&&&& RUNNING data_validation_cumemalloc
buffer size: 327680
off: 0
check 1: MMIO CPU initialization + read back via cuMemcpy D->H
check 2: gdr_copy_to_bar() + read back via cuMemcpy D->H
check 3: gdr_copy_to_bar() + read back via gdr_copy_from_bar()
check 4: gdr_copy_to_bar() + read back via gdr_copy_from_bar() + 5 dwords offset
check 5: gdr_copy_to_bar() + read back via gdr_copy_from_bar() + 11 bytes offset
warning: buffer size 327669 is not dword aligned, ignoring trailing bytes
unmapping
unpinning
&&&& PASSED data_validation_cumemalloc
&&&& RUNNING data_validation_vmmalloc
buffer size: 327680
off: 0
check 1: MMIO CPU initialization + read back via cuMemcpy D->H
check 2: gdr_copy_to_bar() + read back via cuMemcpy D->H
check 3: gdr_copy_to_bar() + read back via gdr_copy_from_bar()
check 4: gdr_copy_to_bar() + read back via gdr_copy_from_bar() + 5 dwords offset
check 5: gdr_copy_to_bar() + read back via gdr_copy_from_bar() + 11 bytes offset
warning: buffer size 327669 is not dword aligned, ignoring trailing bytes
unmapping
unpinning
&&&& PASSED data_validation_vmmalloc
&&&& RUNNING invalidation_access_after_gdr_close_cumemalloc
Mapping bar1
Writing 254 into buf_ptr[0]
Calling gdr_close
Trying to read buf_ptr[0] after gdr_close
Get signal 7 as expected
&&&& PASSED invalidation_access_after_gdr_close_cumemalloc
&&&& RUNNING invalidation_access_after_free_cumemalloc
Mapping bar1
Writing 269 into buf_ptr[0]
Calling gpuMemFree
Trying to read buf_ptr[0] after gpuMemFree
Get signal 7 as expected
&&&& PASSED invalidation_access_after_free_cumemalloc
&&&& RUNNING invalidation_two_mappings_cumemalloc
Mapping bar1
[... 172 lines cut ...]
&&&& PASSED invalidation_unix_sock_shared_fd_gdr_pin_buffer_vmmalloc
&&&& RUNNING invalidation_unix_sock_shared_fd_gdr_map_vmmalloc
parent: Start
child: Start
parent: Calling gdr_open
parent: Calling gdr_pin_buffer
parent: Extracted fd from gdr_t got fd 8
parent: Sending fd to child via unix socket
parent: Extracted gdr_memh_t from gdr_mh_t got handle 0x0
parent: Sending gdr_memh_t to child
parent: Waiting for child to finish
child: Receiving fd from parent via unix socket
child: Got fd 9
child: Converting fd to gdr_t
child: Receiving gdr_memh_t from parent
child: Got handle 0x0
child: Converting gdr_memh_t to gdr_mh_t
child: Attempting gdr_map
child: Cannot do gdr_map as expected
&&&& PASSED invalidation_unix_sock_shared_fd_gdr_map_vmmalloc
92%: Checks: 27, Failures: 2, Errors: 0
sanity.cpp:1856:F:Basic:basic_child_thread_pins_buffer_cumemalloc:0: Failed
sanity.cpp:1862:F:Basic:basic_child_thread_pins_buffer_vmmalloc:0: Failed

@pakmarkthub Here is the output info, thanks!

pakmarkthub, 2022-03-09:
Thank you for the info. What is the gdrdrv version you are using? For libgdrapi, can I ask for confirmation that it is version 2.3?

Based on your post, I guess that you have sudo access to the physical machine. Can you do the followings?
1. On your physical machine, change this line https://github.com/NVIDIA/gdrcopy/blob/master/insmod.sh#L28 to `sudo /sbin/insmod src/gdrdrv/gdrdrv.ko dbg_enabled=1 info_enabled=1`.
2. Do `make driver && sudo ./insmod.sh`.
3. On your container, run `./sanity -v`.
4. Can you post the output from `dmesg` from your **physical machine** here? I want to see the output during the sanity run. You don't need to post the whole log if you don't want to.

myanzhang, 2022-03-09:
@pakmarkthub  Thanks, I will reply when I confirm.