NVIDIA/gdrcopy #232: Does GDRcopy support the HPE/Cray "SlignShot" backbone?

cponder, 2022-08-09:
I'm looking into a performance issue with an app.
If you could tell me up-front whether you support this kind of cluster, it would save some troubleshooting time.

AddyLaddy, 2022-08-09:
Carl, the libfabric plugin and the NCCL plugin have both been able to use GDRCopy on a SlingShot based machine.

See Jim's patch: https://github.com/aws/aws-ofi-nccl/pull/146

cponder, 2022-08-09:
Do you know if UCX can use it? I'll check with the UCX people...

pakmarkthub, 2022-08-10:
Do you mean "can UCX use GDRCopy?" ? I believe that UCX will use GDRCopy if the compile-time options / runtime environments are satisfied. The code is here: https://github.com/openucx/ucx/tree/master/src/uct/cuda/gdr_copy.