NVIDIA/gdrcopy #235: GPUDirect Video

yupeng-ebots, 2022-09-10:
Does this repo works with GPUDirect Video GPU,
more specific I want to use RTX A45000 to load data from FPGA
can we use this repo to transfer data?

Moreover, do we have/plan to have repo for GPUDirect Video?

pakmarkthub, 2022-09-13:
Hi @yupeng-ebots,

GDRCopy is about GPUDirect communication between CPU and GPU. Based on your question, I guess that you want to do GPUDirect between your GPU and FPGA? In that case, you may need to create your own FPGA driver to setup direct communication with the GPU. This document (https://docs.nvidia.com/cuda/gpudirect-rdma/index.html) and the GDRCopy code can be used as a guideline / example.

yupeng-ebots, 2022-09-15:
> Hi @yupeng-ebots,
>
> GDRCopy is about GPUDirect communication between CPU and GPU. Based on your question, I guess that you want to do GPUDirect between your GPU and FPGA? In that case, you may need [...]

Hi Pak,
Thanks so much for your response. I may still have some questions; the document you posted is a feature called GPUDirect RDMA (which is not supported by the latest server GPU, only data center GPU support RDMA at this point). And the feature I am planning to use is GPUDirect Video, which I may not be able to find any tutorial that can start with.
Do we have some related documents?
Best regards.
Yupeng