NVIDIA/gdrcopy #236: Which GDRCopy component needs CUDA as dependency?

roywei, 2022-10-13:
From the instruction it says CUDA is needed for GDRCopy.
`CUDA=<cuda-install-top-dir> ./build-rpm-packages.sh`

I'd like to know if only the library needs CUDA and we can install driver and kernel without CUDA?
Here is our use case:

On cluster/machine level, we only install Nvidia Driver and run the GDR installation, CUDA will not present
```
$ sudo yum groupinstall 'Development Tools'
$ sudo yum install dkms rpm-build make check check-devel subunit subunit-devel
$ cd packages
$ CUDA=<cuda-install-top-dir> ./build-rpm-packages.sh
$ sudo rpm -Uvh gdrcopy-kmod-<version>dkms.noarch.<platform>.rpm
$ sudo rpm -Uvh gdrcopy-<version>.<arch>.<platform>.rpm
$ sudo rpm -Uvh gdrcopy-devel-<version>.noarch.<platform>.rpm

```
On docker, we install CUDA, cudnn, NCCL, and run the GDR installation again.
```
$ sudo yum groupinstall 'Development Tools'
$ sudo yum install dkms rpm-build make check check-devel subunit subunit-devel
$ cd packages
$ CUDA=<cuda-install-top-dir> ./build-rpm-packages.sh
$ sudo rpm -Uvh gdrcopy-kmod-<version>dkms.noarch.<platform>.rpm
$ sudo rpm -Uvh gdrcopy-<version>.<arch>.<platform>.rpm
$ sudo rpm -Uvh gdrcopy-devel-<version>.noarch.<platform>.rpm
```

Will above steps work? Thanks!

pakmarkthub, 2022-10-14:
Hi @roywei,

The test applications, which come with GDRCopy such as `copybw`, require CUDA. GDRCopy driver (gdrdrv) and library (libgdrapi) do not require CUDA. Here are the dependencies.

- gdrcopy-kmod requires NVIDIA driver and dkms.
- gdrcopy-devel requires nothing.
- gdrcopy (contains libgdrapi and the test applications) requires CUDA, check, subunit.

You still need CUDA to run `build-rpm-packages.sh`. It is ok to build those packages on Machine A and install on Machine B.
We also provide some pre-built packages here: https://developer.nvidia.com/gdrcopy (click "Download Now").

roywei, 2022-10-17:
Thank you for the reply!!

closed 2022-10-17 by roywei