NVIDIA/gdrcopy #247: Non-blocking gdr read and write

anaanimous, 2023-02-13:
Can the gdr_copy_to_mapping and gdr_copy_from_mapping operations be made non-blocking? I attempted to utilize cuPointerSetAttribute(&flag, CU_POINTER_ATTRIBUTE_SYNC_MEMOPS, ptr) with a flag value of 0, but it did not change the output of the copylat program.

Furthermore, there is an additional question regarding cudaFlushGPUDirectRDMAWrites(). If the gdr operations are already blocking, what is the purpose of using this function, as presented on slide 38 of the GTC session?

pakmarkthub, 2023-02-14:
Hi @anaanimous,

Can you explain your use case? Depending on that, you can choose to queue up `cudaMemcpy` in a stream or a graph. Although the first call may be expensive, the subsequent calls that have already been queued up can have lower latency. If it is acceptable, you may not need GDRCopy.

In some scenarios such as inside a CUDA Graph host node or from a CUDA host callback function, `cudaMemcpy` cannot be used. In that case, I understand why you need GDRCopy. You can choose to implement non-blocking GDRCopy by having a separate CPU thread doing the work.

In some systems that allow write-combine mappings, `gdr_copy_to_mapping` is partially non-blocking. The data you want to copy to GPU is first copied to the write-combine CPU cache. The API returns without checking that the data has arrived in GPU memory. `cudaFlushGPUDirectRDMAWrites`, launching a new CUDA kernel, to name a few, could flush the on-the-way data to the point of GPU consistency, and thus GPU is safe to consume the data.

GDRCopy, like other RDMA devices, has no knowledge of most of CUDA settings. As you pointed out, setting `cuPointerSetAttribute(&flag, CU_POINTER_ATTRIBUTE_SYNC_MEMOPS, ptr)` has no effect. This also means that GDRCopy side-tracks CUDA memory model, similar to other RDMA devices. You can get more information about how RDMA devices interact with CUDA [here](https://docs.nvidia.com/cuda/gpudirect-rdma/).

anaanimous, 2023-02-14:
Thank you for your reply. Our application operates within a microsecond time scale, and the overhead from the CUDA API can significantly impact performance. The typical size of data transfers in our case is below the threshold where cudaMemcpy outperforms GDR, and we still see benefits in using GDR, even if the transfers are blocking.

In regards to your statement that gdr_copy_to_mapping is partially non-blocking on systems with Write-Combining (WC), could you clarify if this is accurate, given that the code implements write fences? Or is it specific to POWER systems only?

pakmarkthub, 2023-02-15:
> In regards to your statement that gdr_copy_to_mapping is partially non-blocking on systems with Write-Combining (WC), could you clarify if this is accurate, given that the code implements write fences?

To reduce confusion, let me state that `gdr_copy_to_mapping` is designed as a blocking API. But how it works with WC mappings is that it pushes the data into the WC buffer then issues `sfence`. At least for Intel CPU, `sfence` will flush the WC buffer. However, data can be on the way when `sfence` returns -- may still in the PCIe bus, at the GPU incoming port, still being route inside the GPU before reaching the GPU visibility coherency point, etc. That's why `gdr_copy_to_mapping` does not guarantee the visibility nor ordering. Users need to issue `cudaFlushGPUDirectRDMAWrites` to be absolutely sure, especially if write-after-write (WAW) ordering is important. I hope that my explanation clarifies that `gdr_copy_to_mapping` is a blocking API but the meaning of *blocking* might not be as you thought.

anaanimous, 2023-02-15:
Thanks for the clarification.

closed 2023-02-15 by anaanimous