NVIDIA/gdrcopy #253: GDR copy visibility order on GPU

sitecao, 2023-04-19:
Say I have two set of gdr buffers A and B, each with its own memory handle mh_A, mh_B on which host to device copy is performed.

Now, if I do gdr_copy_to_mapping(mh_A,... ) followed by gdr_copy_to_mapping(mh_B, ...), is there a guarantee of the visibility order in the CUDA kernel if it monitors these buffers? If not does something like _mm_sfence() help here?

The usage is that I will prepare data on A to be consumed by GPU and then signal it via B.

pakmarkthub, 2023-04-20:
Hi @sitecao,

If you need the visibility ordering, please use `cuFlushGPUDirectRDMAWrites` (https://docs.nvidia.com/cuda/cuda-driver-api/group__CUDA__DEVICE.html#group__CUDA__DEVICE_1g265e3c82ef0f0fe035f85c4c45a8fbdf). For example,

```
gdr_copy_to_mapping(mh_A, ...);
cuFlushGPUDirectRDMAWrites(...);
gdr_copy_to_mapping(mh_B, ...);
```
It will guarantee that when your CUDA kernel observes B, that kernel can safely consume A.

sitecao, 2023-04-20:
> Hi @sitecao,
>
> If you need the visibility ordering, please use `cuFlushGPUDirectRDMAWrites` [...]

sweet! thank you

closed 2023-04-20 by sitecao