NVIDIA/gdrcopy #254: gdrdrv.ko uses GPL-only symbol 'cc_mkdec'

angainor, 2023-05-04:
When trying to compile `gdrcopy` on Rocky Linux release 9.1, kernel version 5.14.0-162.23.1.el9_1.x86_64 I get this error
```
make[1]: Entering directory '/usr/src/kernels/5.14.0-162.23.1.el9_1.x86_64'
  MODPOST /usr/src/gdrdrv-2.3/Module.symvers
ERROR: modpost: GPL-incompatible module gdrdrv.ko uses GPL-only symbol 'cc_mkdec'
make[2]: *** [scripts/Makefile.modpost:134: /usr/src/gdrdrv-2.3/Module.symvers] Error 1
```
Do you know what I can do to fix this? I have never seen this problem before. Is this related to the kernel version?

Thanks for any help!

pakmarkthub, 2023-05-08:
Hi @angainor,

This issue should have already been fixed in the master branch.

angainor, 2023-05-08:
@pakmarkthub Thanks! I should have checked :) all compiles fine in `master`.