NVIDIA/gdrcopy #262: Ubuntu 22.04 - gdrcopy 2.3.1 dependency problem

YoonGi-AWS, 2023-05-30:
When I try to install gdrcopy 2.3.1 into Ubuntu 22.04(5.19.0-1025-aws), I faced dependency problem.

```
root@ip-172-31-18-103:/home/ubuntu/gdrcopy-2.3.1/packages# dpkg -i gdrdrv-dkms_*_amd64.*.deb
Selecting previously unselected package gdrdrv-dkms:amd64.
(Reading database ... 100387 files and directories currently installed.)
Preparing to unpack gdrdrv-dkms_2.3.1_amd64.Ubuntu22_04.deb ...
Unpacking gdrdrv-dkms:amd64 (2.3.1) ...
Setting up gdrdrv-dkms:amd64 (2.3.1) ...
Loading new gdrdrv-2.3.1 DKMS files...
Building for 5.19.0-1025-aws
Building for architecture x86_64
Building initial module for 5.19.0-1025-aws
EFI variables are not supported on this system
/sys/firmware/efi/efivars not found, aborting.
Done.

gdrdrv.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.19.0-1025-aws/updates/dkms/

depmod...

root@ip-172-31-18-103:/home/ubuntu/gdrcopy-2.3.1/packages# dpkg -i libgdrapi_*_amd64.*.deb
Selecting previously unselected package libgdrapi:amd64.
(Reading database ... 100396 files and directories currently installed.)
Preparing to unpack libgdrapi_2.3.1_amd64.Ubuntu22_04.deb ...
Unpacking libgdrapi:amd64 (2.3.1) ...
Setting up libgdrapi:amd64 (2.3.1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...

root@ip-172-31-18-103:/home/ubuntu/gdrcopy-2.3.1/packages# dpkg -i gdrcopy-tests_*_amd64.*.deb
Selecting previously unselected package gdrcopy-tests:amd64.
(Reading database ... 100404 files and directories currently installed.)
Preparing to unpack gdrcopy-tests_2.3.1_amd64.Ubuntu22_04.deb ...
Unpacking gdrcopy-tests:amd64 (2.3.1) ...
Setting up gdrcopy-tests:amd64 (2.3.1) ...

root@ip-172-31-18-103:/home/ubuntu/gdrcopy-2.3.1/packages# dpkg -i gdrcopy_*_amd64.*.deb
Selecting previously unselected package gdrcopy:amd64.
(Reading database ... 100412 files and directories currently installed.)
Preparing to unpack gdrcopy_2.3.1_amd64.Ubuntu22_04.deb ...
Unpacking gdrcopy:amd64 (2.3.1) ...
dpkg: dependency problems prevent configuration of gdrcopy:amd64:
 gdrcopy:amd64 depends on gdrdrv-dkms (= 2.3.1-1); however:
  Version of gdrdrv-dkms:amd64 on system is 2.3.1.
 gdrcopy:amd64 depends on libgdrapi (= 2.3.1-1); however:
  Version of libgdrapi:amd64 on system is 2.3.1.
 gdrcopy:amd64 depends on gdrcopy-tests (= 2.3.1-1); however:
  Version of gdrcopy-tests:amd64 on system is 2.3.1.

dpkg: error processing package gdrcopy:amd64 (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 gdrcopy:amd64
```

It seems gdrcopy package depend on "2.3.1-1" of  gdrdrv-dkms/libgdrapi/gdrcopy-tests but installed version is 2.3.1.

For the work around, If I change the build-dep-packages.sh file like below, it works.
```
root@ip-172-31-18-103:/home/ubuntu/gdrcopy-2.3.1/packages# diff build-deb-packages.sh.original build-deb-packages.sh 
99,100c99
< FULL_VERSION="${VERSION}-${DEBIAN_VERSION}"
---
> FULL_VERSION="${VERSION}"
```

Could you check this issue?

drossetti, 2023-06-06:
@pakmarkthub can you take a look?

pakmarkthub, 2023-06-20:
Hi @YoonGi-AWS,

Thank you for reporting this issue. The fix is under review.

drossetti, 2023-08-01:
@YoonGi-AWS is the bug fixed?

wenduwan, 2023-08-25:
@pakmarkthub Hi do you plan to produce a new release with the fix?

pakmarkthub, 2023-08-25:
@wenduwan We are working on the next release and will release it soon. For now, you can checkout the latest commit of branch R2.3.1.

wenduwan, 2023-08-25:
@pakmarkthub Thank you! I verified that branch and it works for me on Ubuntu 20.04. Looking forward to the new release.

wenduwan, 2023-09-20:
@pakmarkthub Double check this issue should be resolved with 2.4 release?

pakmarkthub, 2023-09-20:
@wenduwan Yes, it should. Let us know if you still see the issue.

wenduwan, 2023-09-21:
Thank you! I confirmed that it works on Ubuntu 22.04