NVIDIA/gdrcopy #268: Ubuntu 22 - dpkg: error processing package gdrdrv-dkms:amd64 (--install) during installation of gdrcopy

dreambeyondorange, 2023-06-26:
Hi, we are facing an issue that seems similar to https://github.com/NVIDIA/gdrcopy/issues/262 but slightly different and were hoping to get some insights.  We were building gdrcopy on an Ubuntu 22 image from 01/2023 just fine and just recently switched to an image from 06/2023 and starting to see this error when installing gdrcopy

```
Building gdrcopy package ...
       + mkdir -p /tmp/gdr.LKP4bA/gdrcopy-2.3
       + cd /tmp/kitchen/cache/gdrcopy-2.3/packages/..
       + cp -r packages/debian-meta /tmp/gdr.LKP4bA/gdrcopy-2.3/debian
       + cp README.md /tmp/gdr.LKP4bA/gdrcopy-2.3/debian/README.Debian
       + cp README.md /tmp/gdr.LKP4bA/gdrcopy-2.3/debian/README.source
       + cd /tmp/gdr.LKP4bA/gdrcopy-2.3
       + find . -type f -exec sed -i s/@FULL_VERSION@/2.3-1/g {} +
       + find . -type f -exec sed -i s/@VERSION@/2.3/g {} +
       + find . -type f -exec sed -i s/@MODULE_LOCATION@/\/kernel\/drivers\/misc\//g {} +
       + cd /tmp/gdr.LKP4bA
       + tar czvf gdrcopy_2.3.orig.tar.gz gdrcopy-2.3
       gdrcopy-2.3/
       gdrcopy-2.3/debian/
       gdrcopy-2.3/debian/copyright
       gdrcopy-2.3/debian/rules
       gdrcopy-2.3/debian/control
       gdrcopy-2.3/debian/compat
       gdrcopy-2.3/debian/README.source
       gdrcopy-2.3/debian/changelog
       gdrcopy-2.3/debian/source/
       gdrcopy-2.3/debian/source/format
       gdrcopy-2.3/debian/README.Debian
       + debuild -us -uc
        dpkg-buildpackage -us -uc -ui
       dpkg-buildpackage: info: source package gdrcopy
       dpkg-buildpackage: info: source version 2.3-1
       dpkg-buildpackage: info: source distribution stable
       dpkg-buildpackage: info: source changed by Pak Markthub <pmarkthub@nvidia.com>
        dpkg-source --before-build .
       dpkg-buildpackage: info: host architecture amd64
       dpkg-source: warning: unknown information field 'Maintainer' in input data in package's section of control info file
       dpkg-source: warning: unknown information field 'Uploaders' in input data in package's section of control info file
        debian/rules clean
       dh_testdir
       dh_clean
       dh_clean: warning: Compatibility levels before 10 are deprecated (level 9 in use)
        dpkg-source -b .
       dpkg-source: warning: unknown information field 'Maintainer' in input data in package's section of control info file
       dpkg-source: warning: unknown information field 'Uploaders' in input data in package's section of control info file
       dpkg-source: info: using source format '3.0 (quilt)'
       dpkg-source: info: building gdrcopy using existing ./gdrcopy_2.3.orig.tar.gz
       dpkg-source: info: building gdrcopy in gdrcopy_2.3-1.debian.tar.xz
       dpkg-source: info: building gdrcopy in gdrcopy_2.3-1.dsc
        debian/rules build
       make: Nothing to be done for 'build'.
        debian/rules binary
       dh_testdir
       dh_testroot
       dh_prep
       dh_testdir
       dh_testroot
       dh_install
       dh_install: warning: Compatibility levels before 10 are deprecated (level 9 in use)
       dh_installdocs
       dh_installdocs: warning: Compatibility levels before 10 are deprecated (level 9 in use)
       dh_installchangelogs
       dh_compress
       dh_compress: warning: Compatibility levels before 10 are deprecated (level 9 in use)
       dh_fixperms
       dh_installdeb
       dh_installdeb: warning: Compatibility levels before 10 are deprecated (level 9 in use)
       dh_gencontrol
       dpkg-gencontrol: warning: unknown information field 'Maintainer' in input data in package's section of control info file
       dpkg-gencontrol: warning: unknown information field 'Uploaders' in input data in package's section of control info file
       dh_md5sums
       dh_builddeb
       dpkg-deb: building package 'gdrcopy' in '../gdrcopy_2.3-1_amd64.deb'.
        dpkg-genbuildinfo -O../gdrcopy_2.3-1_amd64.buildinfo
        dpkg-genchanges -O../gdrcopy_2.3-1_amd64.changes
       dpkg-genchanges: warning: unknown information field 'Maintainer' in input data in package's section of control info file
       dpkg-genchanges: warning: unknown information field 'Uploaders' in input data in package's section of control info file
       dpkg-genchanges: info: including full source code in upload
        dpkg-source --after-build .
       dpkg-source: warning: unknown information field 'Maintainer' in input data in package's section of control info file
       dpkg-source: warning: unknown information field 'Uploaders' in input data in package's section of control info file
       dpkg-buildpackage: info: full upload (original source is included)
       Now running lintian gdrcopy_2.3-1_amd64.changes ...
       running with root privileges is not recommended!
       E: gdrcopy changes: bad-distribution-in-changes-file stable
       E: gdrcopy: description-starts-with-package-name
       E: gdrcopy source: invalid-standards-version 2.3-1
       W: gdrcopy source: debhelper-but-no-misc-depends gdrcopy
       W: gdrcopy: debian-changelog-has-wrong-day-of-week 2021-07-23 is a Friday
       W: gdrcopy: extended-description-line-too-long line 1
       W: gdrcopy source: no-debian-changes
       W: gdrcopy source: odd-historical-debian-changelog-version 2.2 (for non-native)
       W: gdrcopy source: package-uses-deprecated-debhelper-compat-version 9
       W: gdrcopy source: syntax-error-in-dep5-copyright Continuation line not in paragraph (line 15). Missing a dot on the previous line? [debian/copyright]
       Finished running lintian.

       Copying *.deb and supplementary files to the current working directory ...
       + cd /tmp/kitchen/cache/gdrcopy-2.3/packages
       + cp /tmp/gdr.LKP4bA/gdrcopy_2.3-1_amd64.deb ./gdrcopy_2.3-1_amd64.Ubuntu22_04.deb
       + cp /tmp/gdr.LKP4bA/gdrcopy-tests_2.3-1_amd64.deb ./gdrcopy-tests_2.3-1_amd64.Ubuntu22_04.deb
       + cp /tmp/gdr.LKP4bA/gdrdrv-dkms_2.3-1_amd64.deb ./gdrdrv-dkms_2.3-1_amd64.Ubuntu22_04.deb
       + cp /tmp/gdr.LKP4bA/libgdrapi_2.3-1_amd64.deb ./libgdrapi_2.3-1_amd64.Ubuntu22_04.deb
       + cp /tmp/gdr.LKP4bA/gdrcopy_2.3-1.debian.tar.xz /tmp/gdr.LKP4bA/gdrcopy_2.3.orig.tar.gz /tmp/gdr.LKP4bA/gdrcopy-tests_2.3-1.debian.tar.xz /tmp/gdr.LKP4bA/gdrcopy-tests_2.3.orig.tar.gz /tmp/gdr.LKP4bA/gdrdrv-dkms_2.3-1.debian.tar.xz /tmp/gdr.LKP4bA/gdrdrv-dkms_2.3.orig.tar.gz /tmp/gdr.LKP4bA/libgdrapi_2.3-1.debian.tar.xz /tmp/gdr.LKP4bA/libgdrapi_2.3.orig.tar.gz .
       + cp /tmp/gdr.LKP4bA/gdrcopy_2.3-1.dsc /tmp/gdr.LKP4bA/gdrcopy-tests_2.3-1.dsc /tmp/gdr.LKP4bA/gdrdrv-dkms_2.3-1.dsc /tmp/gdr.LKP4bA/libgdrapi_2.3-1.dsc .

       Cleaning up ...
       + rm -rf /tmp/gdr.LKP4bA
       Selecting previously unselected package gdrdrv-dkms:amd64.
       (Reading database ... 86661 files and directories currently installed.)
       Preparing to unpack gdrdrv-dkms_2.3-1_amd64.Ubuntu22_04.deb ...
       Unpacking gdrdrv-dkms:amd64 (2.3-1) ...
       Setting up gdrdrv-dkms:amd64 (2.3-1) ...
       STDERR: dpkg-buildpackage: warning: using a gain-root-command while being root
        dpkg-source --before-build .
        fakeroot debian/rules clean
       dh_clean: warning: Compatibility levels before 10 are deprecated (level 9 in use)
        dpkg-source -b .
        debian/rules build
        fakeroot debian/rules binary
       dh_installdirs: warning: Compatibility levels before 10 are deprecated (level 9 in use)
       dh_installinit: warning: Compatibility levels before 10 are deprecated (level 9 in use)
       dh_strip: warning: Compatibility levels before 10 are deprecated (level 9 in use)
       dh_compress: warning: Compatibility levels before 10 are deprecated (level 9 in use)
       dh_installdeb: warning: Compatibility levels before 10 are deprecated (level 9 in use)
       dh_shlibdeps: warning: Compatibility levels before 10 are deprecated (level 9 in use)
        dpkg-genbuildinfo --build=full -O../gdrdrv-dkms_2.3-1_amd64.buildinfo
        dpkg-genchanges --build=full -O../gdrdrv-dkms_2.3-1_amd64.changes
        dpkg-source --after-build .
       Loading new gdrdrv-2.3 DKMS files...
       Building for 5.19.0-1027-aws
       Building for architecture x86_64
       Building initial module for 5.19.0-1027-aws
       ERROR (dkms apport): unable to determine source package for gdrdrv-dkms
       Error! Bad return status for module build on kernel: 5.19.0-1027-aws (x86_64)
       Consult /var/lib/dkms/gdrdrv/2.3/build/make.log for more information.
       dpkg: error processing package gdrdrv-dkms:amd64 (--install):
        installed gdrdrv-dkms:amd64 package post-installation script subprocess returned error exit status 10
       Errors were encountered while processing:
        gdrdrv-dkms:amd64
       ---- End output of "bash"  ----
       Ran "bash"  returned 1
```

Is this related to the versioning issue linked above or something different?

dreambeyondorange, 2023-06-26:
Here is our build recipe 
```
def installation_code
  <<~COMMAND
  CUDA=/usr/local/cuda ./build-deb-packages.sh
  dpkg -i gdrdrv-dkms_#{gdrcopy_version_extended}_#{gdrcopy_arch}.#{gdrcopy_platform}.deb
  dpkg -i libgdrapi_#{gdrcopy_version_extended}_#{gdrcopy_arch}.#{gdrcopy_platform}.deb
  dpkg -i gdrcopy-tests_#{gdrcopy_version_extended}_#{gdrcopy_arch}.#{gdrcopy_platform}.deb
  dpkg -i gdrcopy_#{gdrcopy_version_extended}_#{gdrcopy_arch}.#{gdrcopy_platform}.deb
  COMMAND
end
```

pakmarkthub, 2023-06-29:
Hi @dreambeyondorange,

This seems to be unrelated to #262. There seems to be an error when you tried to unpack and build gdrdrv. Can you attach `/var/lib/dkms/gdrdrv/2.3/build/make.log` here? Also, can you compile GDRCopy from source on that machine?

dreambeyondorange, 2023-06-29:
I think this may have been a red herring.  It is no longer failing without intervention

dreambeyondorange, 2023-06-29:
Actually, I am remembering the workaround we did to pin the version of ubuntu 22 to a Jan 6 build.  I will add some logs when I can