NVIDIA/gdrcopy #277: Unable to compile GDRCOPY v2.4

artemry-nv, 2023-09-23:
Build env:
RHEL 7.6 (gcc 4.8.5)
CUDA 12.2.2

Compilation error:
```
# make prefix=${gdrcopy_prefix} CUDA=${CUDA} config lib exes install
GDRAPI_ARCH=X86
cd src && \
make LIB_MAJOR_VER=2 LIB_MINOR_VER=4
make[1]: Entering directory `/tmp/tmp.97ImQVXp1P/gdrcopy/src'
GDRAPI_ARCH=X86
cc -O2 -fPIC -I ../include -I gdrdrv/ -D GDRAPI_ARCH=X86  -c -o gdrapi.o gdrapi.c
cc -O2 -fPIC -I ../include -I gdrdrv/ -D GDRAPI_ARCH=X86  -c -mavx -o memcpy_avx.o memcpy_avx.c
cc -O2 -fPIC -I ../include -I gdrdrv/ -D GDRAPI_ARCH=X86  -c -msse -o memcpy_sse.o memcpy_sse.c
cc -O2 -fPIC -I ../include -I gdrdrv/ -D GDRAPI_ARCH=X86  -c -msse4.1 -o memcpy_sse41.o memcpy_sse41.c
cc -shared -Wl,-soname,libgdrapi.so.2 -o libgdrapi.so.2.4 gdrapi.o memcpy_avx.o memcpy_sse.o memcpy_sse41.o
PATH=/sbin:/usr/sbin:$PATH; ldconfig -n /tmp/tmp.97ImQVXp1P/gdrcopy/src
ln -sf libgdrapi.so.2.4 libgdrapi.so.2
ln -sf libgdrapi.so.2 libgdrapi.so
make[1]: Leaving directory `/tmp/tmp.97ImQVXp1P/gdrcopy/src'
cd tests && \
make CUDA=/hpc/local/oss/cuda12.2.2
make[1]: Entering directory `/tmp/tmp.97ImQVXp1P/gdrcopy/tests'
g++ -O2 -I /hpc/local/oss/cuda12.2.2/include -I ../include -I ../src -I /hpc/local/oss/cuda12.2.2/include  -c -o copybw.o copybw.cpp
g++ -O2 -I /hpc/local/oss/cuda12.2.2/include -I ../include -I ../src -I /hpc/local/oss/cuda12.2.2/include  -c -o common.o common.cpp
g++ -O2 -I /hpc/local/oss/cuda12.2.2/include -I ../include -I ../src -I /hpc/local/oss/cuda12.2.2/include -L /hpc/local/oss/cuda12.2.2/lib64 -L /hpc/local/oss/cuda12.2.2/lib -L /usr/lib64/nvidia -L /usr/lib/nvidia -L /hpc/local/oss/cuda12.2.2/lib64/stubs -L /hpc/local/oss/cuda12.2.2/lib64 -L ../src   -o gdrcopy_copybw copybw.o common.o -lcuda -lpthread -ldl -lgdrapi -lrt
g++ -O2 -I /hpc/local/oss/cuda12.2.2/include -I ../include -I ../src -I /hpc/local/oss/cuda12.2.2/include  -c -o sanity.o sanity.cpp
sanity.cpp: In function ‘int main(int, char**)’:
sanity.cpp:2041:23: error: ‘class std::vector<std::basic_string<char> >’ has no member named ‘emplace_back’
                 tests.emplace_back(optarg);
                       ^
make[1]: *** [sanity.o] Error 1
make[1]: Leaving directory `/tmp/tmp.97ImQVXp1P/gdrcopy/tests'
make: *** [exes] Error 2
```

artemry-nv, 2023-09-23:
Have you dropped RHEL 7.x support in GDRCopy v2.4?

pakmarkthub, 2023-09-23:
Yes, we have dropped RHEL7 support since v2.4. You will need to use v2.3 or earlier instead.

closed 2023-09-23 by artemry-nv